JuliaControl/ControlSystems.jl #440: dare validation unsuitable for H-inf synthesis

kjellqvist, 2021-02-03:
The command dare(A, B, Q, R) has validation on R > 0, this is unsuitable for H-infinity synthesis.

finding the stationary H-inf controller amounts to finding a fixpoint to the dare:

P = Q + A'(P^(-1) + BR^(-1)B -I/gamma^2)A.

This is equivalent to solving the lqr problem with augmented B* = [B I] and R* = [R 0;0 -gamma^2]. The algorithm which is already implemented works well, however R* is indefinite so calling dare(A, B, Q, R) throws an error.

Minimal test case:

gamma = 2
A = ones((1,1))
B = ones((1,2))
Q = ones((1,1))
R = [1. 0;0 -gamma^2]

P = dare(A,B,Q,R)

@test P \approx 1/2 + sqrt(4/3 + 1/4)

olof3, 2021-02-03:
I'm working on a PR fixing this and some other issues, I hope to have the time to finish it any day now.

kjellqvist, 2021-02-03:
Great - could you provide a link to the PR?

olof3, 2021-02-03:
`https://github.com/olof3/ControlMatrixEquations.jl` A lot more that needs to be done and some design decisions that needs to be taken. I'd be happy if you'd give it a try. You can post issues in that package or just let me know directly.

For an idea of the integration with ControlSystems.jl see #443

kjellqvist, 2021-02-04:
Thank you @olof3, I will try it.