JuliaControl/ControlSystems.jl #445: Inconsistent internal notation for sampling time

albheim, 2021-02-16:
Saw there was a mix of `h` and `Ts` mainly in the discrete.jl but some other places also. Nothing major but nice with consistency in my opinion, I can go through and change if there is no specific reason not to. I assume `Ts` is the one we want since that seemed to be used more?

baggepinnen, 2021-02-16:
I might be responsible for some of the mixing, I tend to use h quite often. I do agree that consistency would be to prefer, so please feel free to fix as you see fit, I think Ts has been agreed upon before.

olof3, 2021-02-16:
I also prefer `h` in some circumstances, but as a fieldname or keyword argument I think `Ts` is a lot better. But I agree that  consistency is to be preferred, so feel free to change.

albheim, 2021-02-16:
I agree, `h` comes more natural to me also, but `Ts` is a bit more descriptive so I agree that for the package it is nice.

PR #446 (2021-05-04): Change all sample time to Ts
Fixes #445
closed 2021-05-04 by mfalt