JuliaControl/ControlSystems.jl #450: Error in `damp(sys)` if system is discrete

A-Occ, 2021-02-21:
In the `damp(sys)` function
https://github.com/JuliaControl/ControlSystems.jl/blob/eb0db1ac1a6285dd142b86052d4b241672617c3a/src/analysis.jl#L136-L140

When the system being evaluated is discrete, the bit of code
```
if isdiscrete(sys)
        ps = log(ps)/sys.Ts
 end
```
throws the error `ERROR: MethodError: no method matching log(::Array{Complex{Float64},1})`.

 I believe that this is due to the fact that `ps` is a vector and `log` should be applied element-wise, like
```
if isdiscrete(sys)
        ps = log.(ps)/sys.Ts
 end
```
For the remainder of the function, it seems that everything is being applied element-wise as well, e.g.,
`Wn = abs.(ps)`

albheim, 2021-02-22:
I agree, seems like a simple fix. Maybe a test for `damp` on discrete systems would also be nice to add.

albheim, 2021-02-24:
Thanks for letting us know! I will close this issue now.

closed 2021-02-24 by albheim