JuliaControl/ControlSystems.jl #452: ctrb not working passing A and B matrices as parameters

martinamaggio, 2021-03-02:
This is just to report a problem with the function `ctrb` (running version 0.9.0 on julia version 1.5.3).

When I pass the system as parameter, there is no problem executing the function. When I pass the matrices A and B as parameters I get the following error message

```
julia> ctrb(A,B)
ERROR: MethodError: no method matching ctrb(::Array{Float64,2}, ::Array{Float64,1})
Closest candidates are:
  ctrb(::AbstractArray{T,2} where T, ::AbstractArray{T,2} where T) at
  ~/.julia/packages/ControlSystems/lhpZ5/src/matrix_comps.jl:132
```

Simple code example to reproduce the error.

```
julia> Pkg.status("ControlSystems")
Status `~/.julia/environments/v1.5/Project.toml`
  [a6e380b2] ControlSystems v0.9.0
```

mfalt, 2021-03-02:
The function requires two matrices (Array{Float64,2}) but you have supplied B as a vector (Array{Float64,1}). The ss conductor is nice to automatically handle that and reshape to a matrix, but apparently not ctrb. We should maybe add that.

martinamaggio, 2021-03-02:
Aha, I confirm it works with `reshape`. I think it would definitively be more user-friendly to integrate reshaping (especially because one doesn't need to reshape with `obsv`).

mfalt, 2021-03-02:
> 
> 
> Aha, I confirm it works with `reshape`. I think it would definitively be more user-friendly to integrate reshaping (especially because one doesn't need to reshape with `obsv`).

I don't think we have anything like that implemented for `obsv`, but the difference is that, for example, `C = [1 0]` already is a 1x2 matrix. But, it seems that everyuthing should already work if we allow `B` (or `C` in `obsv`) to be vectors.

However, we should probably not allow `C` to be a vector since they are interpreted as column vectors.

PR #717 (2022-07-12): `ctrb` can take vector `B`
closes #452
closes #453
commit 56849cfd (2022-07-12): close #453
close #452
closed 2022-07-12 by baggepinnen