JuliaControl/ControlSystems.jl #459: Docs not updated by new release

baggepinnen, 2021-03-25:
The links to the "stable" version of the docs corresponding to the latest release
https://juliacontrol.github.io/ControlSystems.jl/stable
has not been updated since v5.3.

albheim, 2021-03-25:
According to documenter it seems like whenever docs are made for a tag it should push to the tag and to stable while otherwise it pushes to dev.

Looking at the github workflow for the docs it seems like it is set up to run for any push to master (creates the dev) and any push to a tag (should create the stable in my mind), but this seems like it is not working.

The [documenter docs](https://juliadocs.github.io/Documenter.jl/stable/man/hosting/#GitHub-Actions) says that tagbot have some problems starting additional workflows if not documenter key is used, which is in line with what we see. But we do have the line for documenter key there so I assume it is also added as a secret and should be working.

What is the workflow for tagging? Write a comment somewhere -> tagbot runs -> should start CI test and doc runs?

baggepinnen, 2021-03-25:
You can create a comment on a commit that mentions `@JuliaRegistrator register` to tag that particular commit, or in an issue to tag the current master commit.

JuliaRegistrator, 2021-03-25:
Error while trying to register: "Tag with name `v0.9.2` already exists and points to a different commit"

albheim, 2021-03-25:
Yeah, on tagbot github page it says 
> For example, if you use GitHub Actions to build the documentation for your package, you will find that with the default TagBot configuration, your documentation build is not triggered when a new tag is created. In this case, you will need to use SSH Deploy Keys.

Under the [deploy keys](https://github.com/JuliaRegistries/TagBot#ssh-deploy-keys) section it is explained how to add a key to allow the tagbots action to trigger additional workflow actions. We have the DOCUMENTER_KEY referenced in our tagbot file,  but is that secret really created and configured correctly? I can't see the secrets so someone else will have to check that.

mfalt, 2021-03-25:
The secrets that I can currently find is ACCESS_TOKEN_BOT for access to the bot that runs tests for plots and a SHA256
"Travis Documenter" "Last used within the last 4 months — Read/write".

I am not sure what kind of key we want, if we want a generic github DeployKey, then I can generate that, but I don't know where we want to add it then, possibly under "Repository secrets"

baggepinnen, 2021-03-25:
Following the steps at
https://juliadocs.github.io/Documenter.jl/stable/man/hosting/#travis-ssh
to generate a deploy key seems to be what I did for MCM.jl a while ago, we only have a "travis documenter" deploy key so that's probably where the problem lies

hyrodium, 2021-07-25:
Maybe this issue is similar to this: https://github.com/JuliaDSP/DSP.jl/issues/434