JuliaControl/ControlSystems.jl #463: Help Request: Time Varying LQR Solution

acxz, 2021-03-31:
It would be great to have a an example which solves systems with time-varying matrices A, B for LQR.

I know I can just call the `lqr` method at every timestep with different A and B matrices, but is that the same?
We use the following, but doesn't that assume that A and B are the same during the backward pass when solving the differential equation concerning the cost to go matrix `X`?
https://github.com/JuliaControl/ControlSystems.jl/blob/311013eef4dd3384872abfca1c04dabb213c97ed/src/matrix_comps.jl#L3-L4

i.e. Solving the cost to go matrix `X` backwards in time with varying `A_t` and `B_t` matrices, instead of constant `A` and `B` matrices.

Maybe I am just not understanding properly LQR in the time-varying case. Any help or guidance would be appreciated. Ideally having an example in the docs would close this issue. Thank you!

olof3, 2021-03-31:
The toolbox only supports the case of infinite horizon and constant system dynamics, which can be solved as an algebraic Riccati equation.

Time-varying LQR is more involved. In continuous time it is necessary to solve a differential Riccati equation. One reference for further reading on this would be Liberzon's book on optimal control, chapter 6.

I think it would be nice to have this functionality in the toolbox at some point.

acxz, 2021-03-31:
Thanks for your comments, something interesting that I just came across is "State-dependent Ricacati Equations" (https://ieeexplore.ieee.org/document/609663).

But yeah the big change would be having an implementation that solves the differential Riccati Equation.

baggepinnen, 2021-04-01:
There is a generalization of this implemented in
https://github.com/baggepinnen/DifferentialDynamicProgramming.jl
I believe The iLQG algorithm will be what you are looking for in case the system is linear and time varying.

closed 2025-01-12 by baggepinnen