JuliaControl/ControlSystems.jl #469: Input/output format of lsim

olof3, 2021-04-27:
In certain ways it is more natural that the matrix-valued (vector-valued time series) input/output data of `lsim` has time along the column dimension, which is the opposite of how it is done now. However, there are some obvious downsides as well. I am leaning towards that we should make this change (and also make the corresponding change to `freqresp`, #321), but would be good to hear thoughts and other pros/cons. @mfalt @baggepinnen @albheim 

Pros:
* States/inputs/output are column vectors, so it is consistent to represent them as columns
* Consistent with memory layout and how we want to do the computations in `lsim`, gets rid of annoying an `Transpose` on the output
* Consistent with how solutions to DifferentialEquations are converted to arrays.
* Avoids the inconsistency that the state `x0` is specified as a (column)-vector but becomes a row vector in the result.

Cons:
* Major breaking change.
* Annoying to view time-series data as row vectors, although this strongly depends on the REPL dimensions (at least Juno inline results are pretty bad).
* Different from how it is done by Matlab and others.

albheim, 2021-04-27:
It seems reasonable to me (without having spent too much time thinking through it). 

Plotting a matrix with Plots.jl it will assume that the different lines are all the columns, so with respect to that it is nice to have time in the column dimension.

murrayrm, 2021-04-27:
Just a note that in `python-control` we opted to put time in the column dimension, as proposed here.  This is handy because it allows you to do plotting in an easier way:
```
t, y = forced_response(sys, u, t)
plt.plot(t, y[0])
```

baggepinnen, 2021-04-27:
> This is handy because it allows you to do plotting in an easier way:

This may be true in python, but Plots.jl expect time to be in the first dimension like we have now.

I do however also think that having time be the last dimension makes sense, I use this convention in most of my other packages for performance reasons. It also makes it free to `reinterpret` the data into a vector of `SVector` since `Vector{SVector{T}}` has the same memory layout as `Matrix{T}`. The vector of vectors interpretation makes a lot of downstream data handling much more convenient than the array representation, so it's nice to be able to convert between them easily.

> Annoying to view time-series data as row vectors

We already have `@autovec` which I guess could partly relieve this problem?

olof3, 2021-04-27:
> Plotting a matrix with Plots.jl it will assume that the different lines are all the columns, so with respect to that it is nice to have time in the column dimension.

Good point, yes that would definitely be annoying.

> We already have @autovec which I guess could partly relieve this problem?

Perhaps when there is a single output? But I guess not in general.

albheim, 2021-05-13:
Switching the time dimension is now implemented in the dev brach with #480

closed 2021-10-21 by baggepinnen