JuliaControl/ControlSystems.jl #471: Missing method Ad, Bd = c2d(A, B, Ts)

albheim, 2021-05-03:
Translating some scripts from matlab to julia I noticed we did not have `Ad, Bd = c2d(A, B, Ts)`. Maybe not very important, but should be easy enough to add.

baggepinnen, 2021-05-03:
Not all c2d methods act only on A and B so feels like it would be limited.

olof3, 2021-05-03:
Yeah, I guess that this could occasionally be useful.

> Not all c2d methods act only on A and B so feels like it would be limited.

Which ones are you thinking about?

baggepinnen, 2021-05-03:
https://github.com/baggepinnen/ControlSystems.jl/blob/4849cb17f48dcdfc89e2d59ede021ec592c2f478/src/discrete.jl#L52

olof3, 2021-05-03:
Seems like you are right and it would be misleading in the case of, e.g., Tustin transformations. For zero-order hold, first-order hold and forward Euler it seems like it could (very) occasionally be useful.

So if we want this, I guess we could just have a wrapper around the existing c2d function with empty C and D matrices and only allow a subset of discretization methods.

albheim, 2021-05-03:
Not important for me, just noticed we did not have it and thought I would bring it up in case it was something that was commonly used since it existed in matlab. But it is not a lot of work to create a state-space and then extract the A and B from the discretized state-space.

olof3, 2021-05-03:
Yes, I think we're better slightly better off not adding any additional code for this rare use case.

It's easy enough to do something like

`Ad, Bd, _, _ = ssdata(c2d(ss(A, B, zeros(0,size(A,1)), 0), 0.1))`

However we are currently not exporting `ssdata`, which I definitely think we should do.

Would probably also be good to add a test to ensure that `c2d` keeps working for `C` matrices with zero rows.

albheim, 2021-05-04:
Closing this, added a PR with your suggestions @olof3 but don't think we need the other stuff

closed 2021-05-04 by albheim