JuliaControl/ControlSystems.jl #478: RFC: Restructuring ControlSystems.jl

ma-laforge, 2021-05-05:
### `ControlSystemsCore.jl`
Looking into wiki page:
-->https://github.com/JuliaControl/ControlSystems.jl/wiki/Restructuring-ControlSystems

I liked the idea of splitting out parts of `ControlSystems.jl`. I didn't think it would be useful to create a separate module for just the types, basic operations & conversions, but I have no objection to this.

### RFC: One more split @ plotting features.
Personally, I would like to see a split of `ControlSystems.jl` at the level where plots are generated.  Maybe it could be called:
 - `ControlSystemsPlots.jl`, or `ControlSystemsAnalysis.jl`

I think this would help `ControlSystems.jl` become leaner, while still keeping the more advanced functions found under the "Analysis", and "Synthesis" umbrella.

My reasonning:
 - Some people might prefer other plotting tools for their own application.
 - The `Plots.jl` environment is relatively dependency heavy, and slow to load.
 - It seems odd that "core" control systems math & types should be tightly coupled to "high-level" plotting routines, anyhow.

albheim, 2021-05-06:
In #235 the plan seems to be to drop the Plots.jl dependency and only keep RecipesBase.jl which is a much lighter dependency. 

Maybe the recipes could still be moved to its own package, but I don't feel it is as needed in that case.

olof3, 2021-05-06:
Thanks you for your comments.

> Looking into wiki page:
-->https://github.com/JuliaControl/ControlSystems.jl/wiki/Restructuring-ControlSystems

That page is quite old and I'm not sure how well it reflects current plans.

As @albheim mentioned, we have discussed breaking out the plotting functionality on a few occasions.  I think it is a good idea and there is a good chance that it might happen relatively soon. I'm closing this issue for now.

closed 2021-05-06 by olof3

albheim, 2021-05-06:
I have missed the earlier discussions. Was the idea to (as in #235) remove the dependency on Plots.jl but still keep the recipes in the package, or completely break out the plotting stuff into its own package? Cause #235 seems like it would not be much work to finish so if that is the route we want to take I could have a quick look at that in the near future.

baggepinnen, 2021-05-06:
I think that as long as we can drop the dependency on Plots we lighten the burden significantly, the recepies are rather cheap.

ma-laforge, 2021-05-06:
@olof3: Excellent!
I was wondering if you were needing help with this task, but it looks like JuliaControl is on the case!

ma-laforge, 2021-05-06:
FWIW: I'm quite ok with just having "plot recipes" as a dependency.

As @baggepinnen says: it should "lighten the burden significantly".