JuliaControl/ControlSystems.jl #479: Problems with baltrunc

olof3, 2021-05-07:
As reported by @ma-laforge in #298, there are some problems with `baltrunc`
```
A = diagm([-1.0, -2, -3])
B = [1 2; 3 4; 5 6]
C = [1 2 3]
D = [1 -1]
sys = ss(A, B, C, D)
```

```
julia> baltrunc(sys)
ERROR: The number of columns of B (2) and D (1) are not equal
```

The problem is the following line
```
if unitgain
    D = D/(C*inv(-A)*B)
end
```

The implementation of `baltrunc` does not seems consistent with the section on model reduction in the given reference (Glad & Ljung), so I am not sure what the fix would be. I guess one solution would be just to update the whole function according to the reference. Do you remember anything about this @baggepinnen

baggepinnen, 2021-05-07:
Hmm, can't remember, seems it was some 5 years ago :P I might have implemented `balreal` from the book but taken a shortcut on `baltrunc`. I have a slightly more general implementation here
https://github.com/JuliaControl/RobustAndOptimalControl.jl/blob/24f974452cf12eaffa7fe0e9d9376edfe5f17ab4/src/reduction.jl#L8
that also supports frequency weights, perhaps we could generalize baltrunc to allow those weights, and default to weights `1`. This implementation does not consider the static gain either, so that has to be added no matter what implementation is used.

baggepinnen, 2021-05-08:
Actually, handling of static gain is a special case of weighted reduction, so it would be handled by having the weight `1/s`

PR #589 (2022-01-04): Fix `baltrunc`
Closes #479
commit 3934d412 (2022-01-04): use residualization in baltrunc

closes #479
closed 2022-01-04 by baggepinnen