JuliaControl/ControlSystems.jl #481: StateSpace to SisoZpk conversion removes states

albheim, 2021-05-10:
Got asked why this was behaving weirdly, and I'm not sure. Seems like the `struct_ctrb_obsv` removes some states when converting from state space to zpk, but was not immediately clear to me what was going on in there.

```julia
using ControlSystems

A = [-0.016303260598174024   0.008606869939999928   0.002945859872611465   0.00038747346072186837
        0.00211965267541959   -0.00211965267541959    0.0                    0.0
        6.776556776556777e-5   0.0                   -6.776556776556777e-5   0.0
        0.0                    0.0                    0.0                   -0.0024333333333333334]
B = [0; 0; 0; 1.0]
C = [1.0 0 0 0]

sys = ss(A, B, C, 0)

@show tzero(sys) pole(sys)
@show tzero(tf(sys)), pole(tf(sys))
@show tzero(zpk(tf(sys))), pole(zpk(tf(sys)))
@show tzero(zpk(sys)), pole(zpk(sys)) # This is different than all the others, still seems similar but some states removed
```

baggepinnen, 2021-05-10:
I've had problems with struct_ctrb_obsv before as well, https://github.com/JuliaControl/ControlSystems.jl/issues/409
maybe it needs a rewrite. I'm not even sure what it does.

olof3, 2021-05-10:
Most recently discussed some days ago in #475

albheim, 2021-05-10:
Okay, closing this then and we can continue discussing in the other ones.

closed 2021-05-10 by albheim