JuliaControl/ControlSystems.jl #503: tf with symbolics does not print correctly

albheim, 2021-05-13:
```julia
julia> L
TransferFunction{Continuous, ControlSystems.SisoRational{Sym}}
            as^2 + bs + 0
-------------------------------------
a*kp + cs^2 + a*ki + b*kp + ds + b*ki

Continuous-time transfer function model

julia> denvec(L)
1×1 Matrix{Vector{Sym}}:
 [a*kp + c, a*ki + b*kp + d, b*ki]
```
Printing transfer function with symbolics looks a bit weird.

olof3, 2021-05-13:
The problem seems to be in Polynomials, see https://github.com/JuliaMath/Polynomials.jl/issues/233

mwe
```julia
using SymPy
using Polynomials
@syms a
Polynomial([a, a^2 + 1])
# prints Polynomial(a + a^2 + 1*x)
```

olof3, 2021-05-13:
Hopefully this just got fixed. Haven't tried it myself though.