JuliaControl/ControlSystems.jl #505: Inconsistencies in lft/feedback/starprod

baggepinnen, 2021-05-14:
It appears as if the implementation of the lft
https://github.com/JuliaControl/ControlSystems.jl/blob/17e78c10ddaa8dbdff672af26eff313e65de904a/src/connections.jl#L269
consistently uses DK22 (s2_D22) instead of DK11 as in the reference. Most upper-left blocks should probably have `s2_D11` instead

@olof3

baggepinnen, 2021-05-14:
There might be some problem with the `s2_C2` element in the lower-right corener of the `A` matrices, should be `s2_C1`

baggepinnen, 2021-05-14:
Actually, it seems that all indices for `s2` are consistently switched `1 <-> 2` making me believe it was a deliberate choice. The book is rather vague on how they define `K`, i.e., is `K` upside down compared to `P` having `y/u` as the upper input-output pair? Could be therefore the indices are switched?

olof3, 2021-05-14:
> Actually, it seems that all indices for s2 are consistently switched 1 <-> 2 making me believe it was a deliberate choice. The book is rather vague on how they define K, i.e., is K upside down compared to P having y/u as the upper input-output pair? Could be therefore the indices are switched?

Yes, exactly. "See Zhou etc. for similar (somewhat less symmetric) formulas." I believe that the modified version that I implemented makes better sense for us.

Should probably write out the reference a bit better, but seems like you had no problem to find it :)

a09174079722A, 2022-01-08:
Ok

closed 2022-01-11 by baggepinnen