JuliaControl/ControlSystems.jl #515: Maybe a mistake to suggest using im for variable

albheim, 2021-05-20:
In the docstring for `nyquist` we suggest using `im` as a variable name. Just thinking this might not be optimal, easy to just copy docstring usage without further thought, and then later in calculation might want to place complex poles. Here it would probably just give an error and some annoyance, but could probably be other scenarios with less obvious problems.

https://github.com/JuliaControl/ControlSystems.jl/blob/17e78c10ddaa8dbdff672af26eff313e65de904a/src/freqresp.jl#L110

baggepinnen, 2021-05-20:
We can get rid of `nyquist` while we make general improvements to `freqresp`, such as 
- accept preallocated output vector
- optimized `logfreqresp`
- threaded `freqresp`
- static computation kernel for small systems

olof3, 2021-05-20:
There is also the list in #321.

A lot of them are quite independent, so no need to them all at once. Getting rid of `nyquist` would be a good start.

closed 2021-06-03 by albheim