JuliaControl/ControlSystems.jl #545: Error with `minreal`

tfoliva, 2021-07-03:
On version v0.10.1. 

If you take 
```
julia> G = tf([1], [1, 3])
TransferFunction{Continuous, ControlSystems.SisoRational{Int64}}
  1
-----
s + 3

Continuous-time transfer function model
 ```
and feedback it with the PI Controller tf
```
julia> CPI = tf([3, 25], [1, 0])
TransferFunction{Continuous, ControlSystems.SisoRational{Int64}}
3s + 25
-------
   s

Continuous-time transfer function model
```

you get
```
julia> CL = feedback(G, CPI)
TransferFunction{Continuous, ControlSystems.SisoRational{Int64}}
      s^2 + 3s
---------------------
s^3 + 9s^2 + 43s + 75

Continuous-time transfer function model
```

Which is not its simplest form. The transfer function has a zero at -3:
```
julia> z
1×1 Matrix{Vector{ComplexF64}}:
 [-3.0 + 0.0im, 0.0 + 0.0im]
```
and a pole at the same location
```
julia> p
1×1 Matrix{Vector{ComplexF64}}:
 [-3.0 + 3.9999999999999982im, -3.0 - 3.9999999999999982im, -3.0 + 0.0im]
```
If you try to use `minreal` to cancel the pole with the zero you get 
```
julia> minreal(CL)
TransferFunction{Continuous, ControlSystems.SisoRational{Float64}}
           1.0s + -0.0
----------------------------------
1.0s^2 + 6.0s + 24.999999999999986

Continuous-time transfer function model
```
Which is wrong. It should have been
```
julia> minreal(CL)
TransferFunction{Continuous, ControlSystems.SisoRational{Float64}}
         1.0
--------------------
1.0s^2 + 6.0s + 25

Continuous-time transfer function model
```
Other than the small rounding difference in the denominator, the main thing is that the result from `minreal` has a zero at 1.

baggepinnen, 2021-07-03:
I don't think this result is wrong, you have a zero in the origin but you have no pole there, to there has to be a zero in the origin after simplification. The result you got does not have a zero in 1, it's in 0.

On the `dev` branch, I get
```julia
julia> G = tf([1], [1, 3])
TransferFunction{Continuous, SisoRational{Int64}}
  1
-----
s + 3

Continuous-time transfer function model

julia> CPI = tf([3, 25], [1, 0])
TransferFunction{Continuous, SisoRational{Int64}}
3s + 25
-------
   s

Continuous-time transfer function model

julia> CL = feedback(G, CPI)
TransferFunction{Continuous, SisoRational{Int64}}
      s
-------------
s^2 + 6s + 25

Continuous-time transfer function model
```

tfoliva, 2021-07-03:
You are right. My mistake, I completely overlooked the zero at the origin. Thanks.

closed 2021-07-03 by tfoliva