JuliaControl/ControlSystems.jl #570: Stable docs not linking correctly?

albheim, 2021-11-15:
Have not looked into how stable is decided, but it links to 0.5.3 which seems old and should probably be updated. Latest docs seems okay.

baggepinnen, 2021-11-15:
It gives some warnings
https://github.com/JuliaControl/ControlSystems.jl/runs/4130341409?check_suite_focus=true#step:7:232
But it doesn't say that it fails with the deployment anywhere

baggepinnen, 2021-11-15:
We appear to be missing this line
https://github.com/baggepinnen/ControlSystemIdentification.jl/blob/ee1f254983e83c3d8d4a936248d81d9697081d4b/.github/workflows/docs.yml#L23

albheim, 2021-11-16:
But why does the latest docs work if that was the problem? Or at least latest it seems to work to me.

baggepinnen, 2021-11-16:
yeah I have no idea :/

ufechner7, 2022-12-15:
This link https://juliacontrol.github.io/ControlSystems.jl/stable is working fine for me and points to 1.5.2 which is correct.

Can this bug be closed?

closed 2022-12-16 by baggepinnen