JuliaControl/ControlSystems.jl #607: MIMO using inconsisten plotting layout between time/freq plots

albheim, 2022-01-21:
For time domain plots it seems like we get one plot per output, and all inputs for each output are sharing that plot.

For freq domain it seems like it is one per input output combination.

Not sure if this has been a conscious choice, or just happened at some point. Don't think it has always been like this, maybe introduced when introducing the `plot(simobject)` syntax for time domain?

baggepinnen, 2022-01-21:
Yep, I think the change was introduced then. Perhaps it's better to have plots `ny x nu` like it was before, it get's hard to see when there are multiple responses in the same pane.

albheim, 2022-01-21:
Consistency over different plot calls is nice, so I think either changing time plots or freq plots would be nice.

I also feel it is more consistent to treat inputs and outputs the same, since it is not obvious to me why you would have this instead of one for each input and collect the corresponding outputs in the same plot? So either all in same plot (maybe messy?) or one plot each. 

But I don't have very strong feelings about it, and I probably won't put time into fixing it myself. Just noticed it and though it seemed odd so put it up here.