JuliaControl/ControlSystems.jl #640: Multiple systems in same plot has inconsistent display

albheim, 2022-02-18:
Have two systems H and H2 to compare. Running
```julia
bodeplot([H, H2])
```
I get a nice plot with legend G1 and G2 so I know which is which. Running
```julia
plot(step.([H, H2]))
```
does not give any. Easy to add in the call, but just felt if there are automatic labels for one one there might as well be for the other to be consistent.

baggepinnen, 2022-02-18:
I'm opposed to automatic labels, because they behave very poorly if you call bodeplot twice. Also, if you plot and set a title, next call to bodeplot will override your title which is quite terrible.

baggepinnen, 2022-02-18:
The same applies to our custom ticks in bode. Plot twice where the axes differ between the plots and the ticks will be messed up.

albheim, 2022-02-18:
I'm mostly arguing for consistency, so I'd be equally up for not having it anywhere (or maybe more now that I think about the problems it might cause).

baggepinnen, 2022-02-18:
I also favor consistency. Every time we've tried to be smart in the plot recipes it has backfired and been prone to breakage with time, so I'd really like to get rid of most of the smartness, like custom ticks, labels and latex stuff.