JuliaControl/ControlSystems.jl #677: How to generate frequency-response data?

ErikQQY, 2022-04-25:
Hello guys! Many thanks for your awesome works! In MATLAB, there is [```frd```](https://www.mathworks.com/help/control/ref/frd.html) to create frequency-response data model. But how to do the same thing in ControlSystems.jl?

baggepinnen, 2022-04-25:
Hey and thanks :) 
There is no `FRD` object in ControlSystems.jl, see [ControlSystemIdentification.FRD](https://baggepinnen.github.io/ControlSystemIdentification.jl/dev/api/#ControlSystemIdentification.FRD)

ErikQQY, 2022-04-25:
Wow! Thanks a lot! That solve my problem! BTW, I think it would be nice to have ```bodeplot``` method for ```FRD``` to generate bodeplot, is there any plan for this? @baggepinnen

ErikQQY, 2022-04-25:
Thank you for your quick response! There are complex numbers in my FRD, so if I directly call plot it would throw errors🥲

baggepinnen, 2022-04-25:
Have you tried? I'm asking because I know it works ;)

```julia
using ControlSystemIdentification, ControlSystems
w = exp10.(LinRange(-2, 2, 200))
r = freqresp(ssrand(1,1,3), w).parent
frd = FRD(w, r)
plot(frd)
```
[image]

ErikQQY, 2022-04-25:
Oops, the old version of ControlSystemIdentification.jl caused the problem.🤣 Anyway, thanks a lot!!

closed 2022-04-25 by ErikQQY

ErikQQY, 2022-04-26:
@baggepinnen Hi there! The plot method of ```FRD``` only generate  the Magnitude-Frequency plot, so I think it would be great to create both Magnitude-Frequency and Phase-Frequency plot like ```bodeplot``` we already have in ControlSystems.jl for FRD type :)

ErikQQY, 2022-04-26:
Besides, I think add ```nyquistplot```, ```margin``` and ```nicholsplot``` method for ```FRD``` type would also be great ;)