JuliaControl/ControlSystems.jl #679: Delays cause `step` and friends to error out.

fedeinthemix, 2022-05-04:
I'm using v.0.12.11 (latest release as of reporting).

The following code produces the expected result
```
let s = tf("s"),
    T = (0.4s^4 + 2.5e6s^3 + 1.9s^2)/(3.4e-14s^6 + 3.7e-7s^5 + 1.4s^4 + 2.5e6s^3 + 1.9s^2),
    fref = 40e6
    plot(step(T/(1 + T)*(1 - delay(2/fref)), 0:0.1/fref:2e-6))
end
```
However, if I increase the maximum simulation time to 5e-6, or increase the delay to 3/fref the procedure errors with
```
BoundsError: attempt to access 241-element Vector{Tuple{Vector{Float64}, Vector{Float64}}} at index [242]

Stacktrace:
  [1] getindex
    @ ./array.jl:805 [inlined]
  [2] _lsim(sys::DelayLtiSystem{Float64, Float64}, u!::Any, t::StepRangeLen{Float64, Base.TwicePrecision{Float64}, Base.TwicePrecision{Float64}}, x0::Vector{Float64}, alg::MethodOfSteps{Tsit5{typeof(OrdinaryDiffEq.trivial_limiter!), typeof(OrdinaryDiffEq.trivial_limiter!), Static.False}, NLFunctional{Rational{Int64}, Rational{Int64}}, false}; kwargs::Base.Iterators.Pairs{Symbol, Float64, Tuple{Symbol, Symbol}, NamedTuple{(:abstol, :reltol), Tuple{Float64, Float64}}})
    @ ControlSystems ~/.julia/packages/ControlSystems/488Bv/src/delay_systems.jl:204
  [3] lsim(sys::DelayLtiSystem{Float64, Float64}, u::ControlSystems.var"#313#314", t::StepRangeLen{Float64, Base.TwicePrecision{Float64}, Base.TwicePrecision{Float64}}; x0::Vector{Float64}, alg::MethodOfSteps{Tsit5{typeof(OrdinaryDiffEq.trivial_limiter!), typeof(OrdinaryDiffEq.trivial_limiter!), Static.False}, NLFunctional{Rational{Int64}, Rational{Int64}}, false}, abstol::Float64, reltol::Float64, kwargs::Base.Iterators.Pairs{Union{}, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
    @ ControlSystems ~/.julia/packages/ControlSystems/488Bv/src/delay_systems.jl:98
  [4] step(sys::DelayLtiSystem{Float64, Float64}, t::StepRangeLen{Float64, Base.TwicePrecision{Float64}, Base.TwicePrecision{Float64}}; kwargs::Base.Iterators.Pairs{Union{}, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
    @ ControlSystems ~/.julia/packages/ControlSystems/488Bv/src/delay_systems.jl:247
  [5] step(sys::DelayLtiSystem{Float64, Float64}, t::StepRangeLen{Float64, Base.TwicePrecision{Float64}, Base.TwicePrecision{Float64}})
    @ ControlSystems ~/.julia/packages/ControlSystems/488Bv/src/delay_systems.jl:240
  [6] top-level scope
    @ ~/tmp/test.ipynb:4
  [7] eval
    @ ./boot.jl:360 [inlined]
  [8] include_string(mapexpr::typeof(REPL.softscope), mod::Module, code::String, filename::String)
    @ Base ./loading.jl:1116
  [9] #invokelatest#2
    @ ./essentials.jl:708 [inlined]
 [10] invokelatest
    @ ./essentials.jl:706 [inlined]
 [11] (::VSCodeServer.var"#146#147"{VSCodeServer.NotebookRunCellArguments, String})()
    @ VSCodeServer /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/packages/VSCodeServer/src/serve_notebook.jl:18
 [12] withpath(f::VSCodeServer.var"#146#147"{VSCodeServer.NotebookRunCellArguments, String}, path::String)
    @ VSCodeServer /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/packages/VSCodeServer/src/repl.jl:185
 [13] notebook_runcell_request(conn::VSCodeServer.JSONRPC.JSONRPCEndpoint{Base.PipeEndpoint, Base.PipeEndpoint}, params::VSCodeServer.NotebookRunCellArguments)
    @ VSCodeServer /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/packages/VSCodeServer/src/serve_notebook.jl:14
 [14] dispatch_msg(x::VSCodeServer.JSONRPC.JSONRPCEndpoint{Base.PipeEndpoint, Base.PipeEndpoint}, dispatcher::VSCodeServer.JSONRPC.MsgDispatcher, msg::Dict{String, Any})
    @ VSCodeServer.JSONRPC /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/packages/JSONRPC/src/typed.jl:67
 [15] serve_notebook(pipename::String; crashreporting_pipename::String)
    @ VSCodeServer /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/packages/VSCodeServer/src/serve_notebook.jl:94
 [16] top-level scope
    @ /nix/store/ysjj70d8gmdssv6ldmpb43isbifyk59j-vscode-extensions/share/vscode/extensions/julialang.language-julia/scripts/notebook/notebook.jl:12
 [17] include(mod::Module, _path::String)
    @ Base ./Base.jl:386
 [18] exec_options(opts::Base.JLOptions)
    @ Base ./client.jl:285
 [19] _start()
    @ Base ./client.jl:485
```

It looks like the size of some internal vector hasn't the expected size.

baggepinnen, 2022-05-04:
The problem appears to be in the ODE solver, the warning
```
┌ Warning: dt <= dtmin. Aborting. There is either an error in your model specification or the true solution is unstable.
└ @ SciMLBase ~/.julia/packages/SciMLBase/GW7GW/src/integrator_interface.jl:345
```
is an indication that the solver does not like the numerics of the problem. One possible way of approaching this would be to rescale the time axis to get coefficients closer to 1. For example by changing the time unit to microseconds instead of seconds.

baggepinnen, 2022-05-04:
Another option is to solve in with higher precision, the following makes the system have `BigFloat` coefficients and solves without problems
```julia
using ControlSystems
s = tf("s")
T = (0.4s^4 + 2.5e6s^3 + 1.9s^2)/(big(3.4e-14)*s^6 + 3.7e-7s^5 + 1.4s^4 + 2.5e6s^3 + 1.9s^2)
T = ss(T)
fref = 40e6
sys = T/(1 + T)*(1 - delay(2/fref))
sol = step(sys, 0:0.1/fref:5e-6)
plot(sol)
```

baggepinnen, 2022-05-04:
By the way, the time scales represented by this model differ by many orders of magnitude
```
julia> poles(T)
6-element Vector{ComplexF64}:
                   0.0 + 0.0im
                   0.0 + 0.0im
  -6.115098457052054e6 + 0.0im
  -2.383627242061831e6 + 2.518444206030512e6im
  -2.383627242061831e6 - 2.518444206030512e6im
 -7.600393276776295e-7 + 0.0im
```
and the model appears to be nonminimal, there's both a pole and a zero in `-7.600393276776295e-7 + 0.0im`

baggepinnen, 2022-05-04:
Yet another point, `T/(1 + T)` is better written as `feedback(T)` to get a minimal representation.

baggepinnen, 2022-05-04:
The following simulates fine without BigFloat
I have
1. manually removed the two poles and zeros in the origin by lowering the degree of the polynomials by two
2. Performed `minreal` to remove the last pole-zero cancellation
3. verify that frequency-responses look similar
4. give `force_dtmin=true` to the solver to ignore the small step size

```julia
using ControlSystems, RobustAndOptimalControl
s = tf("s")
T = (0.4s^2 + 2.5e6s^1 + 1.9)/((3.4e-14)*s^4 + 3.7e-7s^3 + 1.4s^2 + 2.5e6s^1 + 1.9)
# T = (0.4s^4 + 2.5e6s^3 + 1.9s^2)/(3.4e-14s^6 + 3.7e-7s^5 + 1.4s^4 + 2.5e6s^3 + 1.9s^2)
T = ss(T)
Tr = minreal(T)

w = exp10.(LinRange(1, 8, 2000))
bodeplot([T, Tr], w)

fref = 40e6
sys0 = feedback(Tr)
sys = sys0*(1 - delay(2/fref))
sol = step(sys, 0:0.1/fref:15e-6, force_dtmin=true)
plot(sol)
```

fedeinthemix, 2022-05-04:
The system is indeed 3rd order and using `T/(1+T)` I didn't realize that it wasn't simplified. Using `feedback` is definitely useful for that. Using `force_dtmin` works, but I agree that re-scaling time is the way to go. The top error message made it appear a vector size problem and not an integration problem.

Thanks for your help!

closed 2022-05-04 by fedeinthemix

fedeinthemix, 2022-05-04:
By the way, I don't see the Warning. Is it because I'm in a Jupyter notebook, or are you enabling some debugging?

baggepinnen, 2022-05-04:
It might be due to Jupyter, it appears in a standard terminal without setting any flags. 

We don't do any simplification automatically because it requires you to select a numerical tolerance. When calling `minreal`, this tolerance can be provided by the user if the default tolerance is not appropriate, but for operations like `T/(1+T)` this is not possible.