JuliaControl/ControlSystems.jl #740: Discontinuous Root Locus

agerlach, 2022-09-06:
I expect the issue is originating in the sorting in `getpoles`.

```julia
using Plots, ControlSystems
Nroots = -1.0 .+  [-1.732050807568877im, 1.732050807568877im]
Droots =  [0.0, -4.0, -6.0, -0.7 - 0.7141428428542851im, -0.7 + 0.7141428428542851im]
G = zpk(Nroots, Droots, 1.0)
rlocusplot(G, 200)
```

[image]

baggepinnen, 2022-09-07:
Thanks for the example!

Fixed in https://github.com/JuliaControl/ControlSystems.jl/pull/741

[image]

closed 2022-09-07 by baggepinnen

agerlach, 2022-09-07:
No problem. Thanks for the quick fix!

PR #741 (2022-09-07): Fix rlocus when poles cross in both real and imag
Uses the Hungarian algorithm to find the optimal assignment between the new roots and the roots form the previous iteration. This is more robust than sorting based on either imag or real value.

Closes #740

[image]
commit 3f8071b8 (2022-09-07): Fix rlocus when poles cross in both real and imag (#741)

* Fix rlocus when poles cross in both real and imag

Uses the Hungarian algorithm to find the optimal assignment between the new roots and the roots form the previous iteration.

Closes #740

* add compat bound