JuliaControl/ControlSystems.jl #762: Delay system in feedback path throws a method error

jamestjsp, 2022-10-28:
The following code throws a method error. This can be easily reproduced by placing a delay system in the denominator of any expression.
```
L = 5; # Delay time
P = 10/(10*s + 1) * delay(L);
Kp= 0.1;
Ti= 10;
C = Kp*(1 + 1/(Ti*s));
T = C*P/(1+C*P);
t =100;
plot(step(T,t))
```
**Work around.**
Use: `T = feedback(C*P, 1); `

_MethodError: /(::DelayLtiSystem{Float64, Float64}, ::DelayLtiSystem{Float64, Float64}) is ambiguous. Candidates:
  /(anything, sys::DelayLtiSystem) in ControlSystemsBase at ~\.julia\packages\ControlSystemsBase\CKedD\src\types\DelayLtiSystem.jl:93
  /(sys1::LTISystem, sys2::LTISystem) in ControlSystemsBase at ~\.julia\packages\ControlSystemsBase\CKedD\src\types\Lti.jl:5
Possible fix, define
  /(::LTISystem, ::DelayLtiSystem)

Stacktrace:
 [1] top-level scope
   @ In[16]:7
 [2] eval
   @ .\boot.jl:368 [inlined]
 [3] include_string(mapexpr::typeof(REPL.softscope), mod::Module, code::String, filename::String)
   @ Base .\loading.jl:1428_

baggepinnen, 2022-10-28:
Thanks for the report! We do not support inverting delay systems so you will have to use the `feedback` function for this, but we should try to improve the error message to indicate this.

baggepinnen, 2022-10-28:
In general, using hte `feedback` function is always recommended even for cases when we do support inversion, the inversion will not give you a minimal realization and you'll thus have to rely on numerical approximation through `minreal` to obtain a minimal realization after forming the feedback loop. The `feedback` function creates the minimal realizatoin directly.

closed 2025-01-12 by baggepinnen