JuliaControl/ControlSystems.jl #767: lsim, error message for single input system

wahlquisty, 2022-11-02:
If I want to simulate a system with one input, for example
```
using ControlSystemsBase
A = ones(3,3);
B = ones(3,1);
C = ones(1,3);
D = 0;
model = ss(A,B,C,D);

t = LinRange(0,10,100) |> Vector
u = [ones(10,1); zeros(90,1)];
modeld = c2d(model,t[2]-t[1]);
```
and I want to call ``` lsim(modeld,u,t) ```. However, this results in the error message ```u must be of size (nu, length(t))``` and the input `u` needs to be transposed to be a row vector ``` lsim(modeld,u',t) ```.
I suggest either adding the special case of ```nu == 1``` or improve the error message to clarify.

mzaffalon, 2022-11-02:
I have made a PR for the continuous case, https://github.com/JuliaControl/ControlSystems.jl/pull/761 but it is worth updating it for the discrete one too.

mzaffalon, 2022-11-10:
This can be closed now. With v1.5.1 I get
```
julia> lsim(modeld,u,t)
ERROR: u must have size (1, 100): got size (100, 1)
```

baggepinnen, 2022-11-10:
awesome, thanks! :)

closed 2022-11-10 by baggepinnen