JuliaControl/ControlSystems.jl #773: Crating discreate delay system is not allowed as delay is continues

jamestjsp, 2022-11-17:
when delay block is multiplied with a discreet Transferfunction the following error will be thrown. 
ERROR: MethodError: no method matching Continuous(::Discrete{Int64})

Workaround:
Create a discreet  delay using c2d(delay(L), Td) then multiply with discreet Transferfunction.

It is good to have this done automatically.

albheim, 2022-11-17:
Since `delay` creates a `DelayLtiSystem` this suggestion would mean a discrete system multiplied by any general `DelayLtiSystem` would result in an automatic discretization. 

I feel like discretizing general systems automatically is maybe not a good idea and should be something that is explicitly done by the programmer to avoid mistakes, thus my preference would be that it errors, but maybe the error message could be improved a bit. Though I can't say my stake in this is big, so don't listen too much to me.

baggepinnen, 2022-11-17:
> It is good to have this done automatically.

I do not agree, continuous time systems and discrete-time systems are fundamentally different, and implicit conversion is a foot gun that we'd rather avoid. I thus agree with Albin that we should keep discretization an explicit action by the user.

#774 instead adds a discrete-time constructor for `delay` to make this smoother! Thanks for the issue :)

PR #774 (2022-11-17): add discrete-time delay constructor
closes #773
commit c0320a0a (2022-11-17): add discrete-time delay constructor

closes #773
closed 2022-11-17 by baggepinnen