JuliaControl/ControlSystems.jl #785: rlocusplot shows a very long warning with Plots

ufechner7, 2022-12-15:
Example:
```julia
using ControlSystems, Plots
sys = tf([3, 1], [9, 7, 5, 6])
rlocusplot(sys, c=:blue)
```
```
(control) pkg> st
Status `~/repos/control/Project.toml`
  [a6e380b2] ControlSystems v1.5.2
  [91a5bcdd] Plots v1.37.2
```
Output:
```
julia> rlocusplot(sys, c=:blue) # colors are specified with the c or color keyword
┌ Warning: Keyword argument hover not supported with Plots.GRBackend().  Choose from: annotations, arrow, aspect_ratio, background_color, background_color_inside, background_color_outside, background_color_subplot, bar_width, bins, bottom_margin, camera, clims, color_palette, colorbar, colorbar_entry, colorbar_scale, colorbar_title, connections, contour_labels, discrete_values, fill_z, fillalpha, fillcolor, fillrange, fillstyle, flip, foreground_color, foreground_color_axis, foreground_color_border, foreground_color_grid, foreground_color_subplot, foreground_color_text, framestyle, grid, gridalpha, gridlinewidth, gridstyle, group, guide, guidefontcolor, guidefontfamily, guidefonthalign, guidefontrotation, guidefontsize, guidefontvalign, html_output_format, inset_subplots, label, layout, left_margin, legend_background_color, legend_font_color, legend_font_family, legend_font_halign, legend_font_pointsize, legend_font_rotation, legend_font_valign, legend_foreground_color, legend_position, legend_title, levels, lims, line_z, linealpha, linecolor, linestyle, linewidth, link, margin, marker_z, markeralpha, markercolor, markershape, markersize, markerstrokealpha, markerstrokecolor, markerstrokewidth, normalize, orientation, overwrite_figure, permute, polar, primary, projection, quiver, ribbon, right_margin, scale, series_annotations, seriesalpha, seriescolor, seriestype, show, show_empty_bins, size, smooth, subplot, subplot_index, tick_direction, tickfontcolor, tickfontfamily, tickfonthalign, tickfontrotation, tickfontsize, tickfontvalign, ticks, title, titlefontcolor, titlefontfamily, titlefonthalign, titlefontrotation, titlefontsize, titlefontvalign, top_margin, weights, widen, window_title, x, xdiscrete_values, xerror, xflip, xforeground_color_axis, xforeground_color_border, xforeground_color_grid, xforeground_color_text, xgrid, xgridalpha, xgridlinewidth, xgridstyle, xguide, xguidefontcolor, xguidefontfamily, xguidefonthalign, xguidefontrotation, xguidefontsize, xguidefontvalign, xlims, xlink, xscale, xtick_direction, xtickfontcolor, xtickfontfamily, xtickfonthalign, xtickfontrotation, xtickfontsize, xtickfontvalign, xticks, xwiden, y, ydiscrete_values, yerror, yflip, yforeground_color_axis, yforeground_color_border, yforeground_color_grid, yforeground_color_text, ygrid, ygridalpha, ygridlinewidth, ygridstyle, yguide, yguidefontcolor, yguidefontfamily, yguidefonthalign, yguidefontrotation, yguidefontsize, yguidefontvalign, ylims, ylink, yscale, ytick_direction, ytickfontcolor, ytickfontfamily, ytickfonthalign, ytickfontrotation, ytickfontsize, ytickfontvalign, yticks, ywiden, z, z_order, zdiscrete_values, zerror, zflip, zforeground_color_axis, zforeground_color_border, zforeground_color_grid, zforeground_color_text, zgrid, zgridalpha, zgridlinewidth, zgridstyle, zguide, zguidefontcolor, zguidefontfamily, zguidefonthalign, zguidefontrotation, zguidefontsize, zguidefontvalign, zlims, zlink, zscale, ztick_direction, ztickfontcolor, ztickfontfamily, ztickfonthalign, ztickfontrotation, ztickfontsize, ztickfontvalign, zticks, zwiden
└ @ Plots ~/.julia/packages/Plots/Hxe7H/src/args.jl:1553

```

baggepinnen, 2022-12-16:
The warning is due to the plot recipe having hover information, but you’re plotting the GR, a backend that does not support hover information. If you use plotly, you can hover the mouse above the plot to see the gain value at a particular point on the curve.

ufechner7, 2022-12-16:
Well, I guess it could be fixed in this package by using only keywords that are supported by the current backend.

mzaffalon, 2022-12-16:
I would be against adding ad-hoc code only because it is not yet fixed in Plots. Moreover the warning appears only once.

closed 2022-12-19 by baggepinnen