JuliaControl/ControlSystems.jl #787: bodeplot showing nothing after 10^2

urdutext, 2022-12-24:
ControlSystems v1.5.3

using ControlSystems
setPlotScale("dB")
sysc1 = tf(1000000000,[47,0])
fig= bodeplot(sysc1, plotphase=false,label="Capacitor-1",xlims=(1, 10^4),ylims=(-150, 150))

[image]

baggepinnen, 2022-12-24:
That looks like an issue with Plots.jl or GR, have you tried plotting with plotly as backend?

urdutext, 2022-12-25:
> That looks like an issue with Plots.jl or GR, have you tried plotting with plotly as backend?
I have plotted  "using Plots" with gr() 

using ControlSystems
using Plots
gr()
setPlotScale("dB")
sysc1 = tf(1000000000,[47,0])
fig= bodeplot(sysc1, plotphase=true,label="Capacitor-1",xlims=(1, 10^4),ylims=(-150, 150))

baggepinnen, 2022-12-25:
Install Plotly.jl and then call `plotly()`

albheim, 2022-12-25:
Isn't it just that you change xlims without increasing the frequency range? I think there is a `w` keyword for bodeplot allowing you to do that, though not at a computer now so haven't tried.

urdutext, 2022-12-25:
> Isn't it just that you change xlims without increasing the frequency range? I think there is a `w` keyword for bodeplot allowing you to do that, though not at a computer now so haven't tried.

so it is applying default frequency vector by calling  _default_freq_vector which is using default range from 10^0 to 10^2
Thanks!

following code is doing the job.

using ControlSystems
using Plots
gr()
setPlotScale("dB")
nw = round(Int, max(200, 60*(7 - 0)))
# frequency vector
fv= exp10.(range(0, stop=7, length=nw))
sysc1 = tf(100000000,[47,0])
fig= bodeplot(sysc1, plotphase=false,fv,label="Capacitor-1",xlims=(1, 10^7),ylims=(-150, 150))

closed 2022-12-25 by baggepinnen