JuliaControl/ControlSystems.jl #798: marginplot + setPlotScale("dB") = no magnitude plot

stephans3, 2023-01-30:
Hi all,
I tried to use marginplot with setPlotScale("dB").

With log10 it works, but not with dB.

```julia 
using ControlSystems, Plots

G = tf([100], [1, 5, 6, 5, 1])  # Example TF
marginplot(G) # This works

setPlotScale("dB")
marginplot(G) # Does not work
```

These warnings show up:
```
┌ Warning: No strict ticks found
└ @ PlotUtils ~/.julia/packages/PlotUtils/M092a/src/ticks.jl:191
┌ Warning: No strict ticks found
└ @ PlotUtils ~/.julia/packages/PlotUtils/M092a/src/ticks.jl:191
```
[image]

albheim, 2023-01-30:
Seems like a few spots in `ControlSystemsBase/plotting.jl` are not really using the internal `_PlotScale****` variables. Might be enough to update this line
https://github.com/JuliaControl/ControlSystems.jl/blob/77e2eaaabf56a1f3f24271f60cc58ec47517bb17/lib/ControlSystemsBase/src/plotting.jl#L274
to use `_PlotScaleFunc` instead of `:log10`, but might also be more than that needed. And the label just next to that should probably also be updated to use `_PlotScaleStr` in some way as well.

baggepinnen, 2023-01-31:
Hopefully fixed by
- #799

commit 3018ea60 (2023-01-31): fix #798
PR #799 (2023-01-31): Use `yscale --> _PlotScaleFunc` in some plot recipes
closes #798
closed 2023-01-31 by baggepinnen