JuliaControl/ControlSystems.jl #812: ControlSystemIdentification and ControlSystems failed to precompile in Julia 1.8.5

jamestjsp, 2023-02-27:
Version 1.8.5 (2023-01-08)
Built by Homebrew (v1.8.5)
```
Precompiling project...
  ✗ LowLevelParticleFilters
  ✗ ControlSystemIdentification
  ✗ LinearSolve
  ✗ NonlinearSolve
  ✗ OrdinaryDiffEq
  ✗ DelayDiffEq
  ✗ ControlSystems
  39 dependencies successfully precompiled in 99 seconds. 262 already precompiled.
  7 dependencies errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the packages
```

jamestjsp, 2023-02-27:
Package import error trace
```

julia> using ControlSystemIdentification, ControlSystemsBase
[ Info: Precompiling ControlSystemIdentification [3abffc1c-5106-53b7-b354-a47bfc086282]
ERROR: LoadError: error in method definition: function StatsAPI.weights must be explicitly imported to be extended
Stacktrace:
 [1] top-level scope
   @ none:0
 [2] top-level scope
   @ ~/.julia/packages/LowLevelParticleFilters/wDxrV/src/PFtypes.jl:226
 [3] include(mod::Module, _path::String)
   @ Base ./Base.jl:419
 [4] include(x::String)
   @ LowLevelParticleFilters ~/.julia/packages/LowLevelParticleFilters/wDxrV/src/LowLevelParticleFilters.jl:1
 [5] top-level scope
   @ ~/.julia/packages/LowLevelParticleFilters/wDxrV/src/LowLevelParticleFilters.jl:26
 [6] include
   @ ./Base.jl:419 [inlined]
 [7] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::String)
   @ Base ./loading.jl:1554
 [8] top-level scope
   @ stdin:1
in expression starting at /Users/<>/.julia/packages/LowLevelParticleFilters/wDxrV/src/PFtypes.jl:226
in expression starting at /Users/<>/.julia/packages/LowLevelParticleFilters/wDxrV/src/LowLevelParticleFilters.jl:1
in expression starting at stdin:1
ERROR: LoadError: Failed to precompile LowLevelParticleFilters [d9d29d28-c116-5dba-9239-57a5fe23875b] to /Users/<>/.julia/compiled/v1.8/LowLevelParticleFilters/jl_auqN3y.
Stacktrace:
  [1] error(s::String)
    @ Base ./error.jl:35
  [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::IO, internal_stdout::IO, keep_loaded_modules::Bool)
    @ Base ./loading.jl:1707
  [3] compilecache
    @ ./loading.jl:1651 [inlined]
  [4] _require(pkg::Base.PkgId)
    @ Base ./loading.jl:1337
  [5] _require_prelocked(uuidkey::Base.PkgId)
    @ Base ./loading.jl:1200
  [6] macro expansion
    @ ./loading.jl:1180 [inlined]
  [7] macro expansion
    @ ./lock.jl:223 [inlined]
  [8] require(into::Module, mod::Symbol)
    @ Base ./loading.jl:1144
  [9] include
    @ ./Base.jl:419 [inlined]
 [10] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::Nothing)
    @ Base ./loading.jl:1554
 [11] top-level scope
    @ stdin:1
in expression starting at /Users/<>/.julia/packages/ControlSystemIdentification/iYHcD/src/ControlSystemIdentification.jl:1
in expression starting at stdin:1
ERROR: Failed to precompile ControlSystemIdentification [3abffc1c-5106-53b7-b354-a47bfc086282] to /Users/<>/.julia/compiled/v1.8/ControlSystemIdentification/jl_Gooc0j.
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:35
 [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::IO, internal_stdout::IO, keep_loaded_modules::Bool)
   @ Base ./loading.jl:1707
 [3] compilecache
   @ ./loading.jl:1651 [inlined]
 [4] _require(pkg::Base.PkgId)
   @ Base ./loading.jl:1337
 [5] _require_prelocked(uuidkey::Base.PkgId)
   @ Base ./loading.jl:1200
 [6] macro expansion
   @ ./loading.jl:1180 [inlined]
 [7] macro expansion
   @ ./lock.jl:223 [inlined]
 [8] require(into::Module, mod::Symbol)
   @ Base ./loading.jl:1144

```

baggepinnen, 2023-02-27:
Hello and thanks for the issue. The issue appears to be with LowLevelParticleFilters.jl rather than ControlSystems.jl, it's fixed by https://github.com/baggepinnen/LowLevelParticleFilters.jl/pull/97

closed 2023-02-27 by baggepinnen