JuliaControl/ControlSystems.jl #815: `feedback_control` causes a stackoverflow on DelayLtiSystem

pepijndevos, 2023-03-05:
```
julia> sys = feedback_control(tf([1], [0.1, 1]), delay(0.1))
ERROR: StackOverflowError:
Stacktrace:
 [1] feedback(sys1::DelayLtiSystem{Float64, Float64}, sys2::DelayLtiSystem{Float64, Float64}; kwargs::Base.Pairs{Symbol, Any, Tuple{Symbol, Symbol}, NamedTuple{(:Z2, :Zperm), Tuple{Colon, Vector{Int64}}}}) (repeats 39992 times)
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/8Sk6N/src/types/Lti.jl:34
```

baggepinnen, 2023-03-06:
I opened #817 to keep track of making this work. I'd like to solve a more general problem rather than adding a rather complex hack to support this particular instance of the problem, unfortunately, the general problem requires quite a bit of work to implement.

baggepinnen, 2023-03-06:
For now, you can obtain the closed loop systems from `r->y` and `r->u` separately using 
```julia
julia> Gry = feedback(tf([1], [0.1, 1]) * delay(0.1));

julia> Gru = feedback(delay(0.1), tf([1], [0.1, 1]));
```

PR #816 (2023-03-06): prevent stackoverflow on feedback with kwargs
closes #815
commit d1ef7f00 (2023-03-06): prevent stackoverflow on feedback with kwargs

closes #815
closed 2023-03-06 by baggepinnen