JuliaControl/ControlSystems.jl #834: Julia v1.9: ControlSystems/-MTK error message

B-LIE, 2023-05-11:
The following command worked in Julia v1.8.5, but gives an error message in Julia v1.9/latest update of ControlSystems packages (CS: v1.7.2, CSMTK: v0.1.12):
[image]

B-LIE, 2023-05-11:
Hm... the problem seems to be with `minreal`. If I remove piping the linearized model to `minreal`, the code runs, and I get the following bode plot:
[image]

When I *included* the call to `minreal` (Julia v1.8.5), the result was a somewhat different `bodeplot` (which is strange...):
[image]

albheim, 2023-05-11:
How does it look without `minreal` on 1.8.5, is it just `minreal` that does the difference or has something changed with newer versions? 
It seems `minreal` takes some tolerance, so I assume it could simplify away some states that does not exactly cancel depending on the tolerance, though I don't really have a clue.

Can you run `minreal` on a normal state space from ControlSystems.jl, e.g. `minreal(ssrand(1, 1, 3))`, or do you get the same error there?

mzaffalon, 2023-05-11:
> Not a good excuse for not learning about projects and toml, I know...

It is as easy as creating a new directory, starting the Julia version you want, hitting "]" so that the prompt changes to `pkg>` and type `add ControlSystems`. You can pin a ControlSystems version, e.g. v1.7.0, by using `pin ControlSystems@v1.7.0`.

Could you please paste the code (in a block of three backticks) instead of making a screenshot?

B-LIE, 2023-05-11:
OK... part of my "problem" with installing various versions of Julia packages is the following...
* The documentation is good on doing this in the documentation of `Pkg.jl`, i.e., to manage projects.
* I don't know where to find documentation on how to specify the *Julia* version -- the `toml`?

The reason why this is important: 1-2 years ago, I updated to a new version of Julia, and then it took several months before package `Turing.jl` worked with the new Julia version. It is not realistic that a user *remembers* which Julia version to use for a specific package version, so it is important to be able to package both the specification of the package version number and the Julia version number. 

And... I don't know where the documentation for specifying the Julia version is to be found...
##
OK -- here is the code I used:
``` julia
sys_test = ss([-1 2 3; 0 -2 3; 0 0 -3], [1;2;3], [1 2 3], [0])

minreal(sys_test)
```

mzaffalon, 2023-05-11:
Maybe [here](https://pkgdocs.julialang.org/v1/compatibility/) and using in the `Project.toml` file

```
[compat]
julia = "=1.8.5"
```
?

baggepinnen, 2023-05-11:
The warnings during precompilation are unrelated. What system is you on? If MacOS, are you running one of their newer ARM processors? The code works on my machine under both Julia 1.8.5 and 1.9.

`minreal` is implemented in MatrixPencils.jl and, as Albin says, takes a number of tolerance arguments that can be tweaked to get your desired results.

baggepinnen, 2023-05-11:
Could you try deleting your entire `.julia/compiled` folder and then issue `pkg> precompile` to re-do all precompilation? Your compilation cache might have ended up in a bad state or something.

B-LIE, 2023-05-11:
OK -- 
[image]

So 
1. I select subfolder `compiled` under directory `.julia` and delete it, and 
2. Run the `precompile` command in the `Pkg` package.

I'll do that. (I'll rename the `compiled` folder in case I get a problem...)

B-LIE, 2023-05-11:
OK -- I have recompiled. Here are the packages that I have installed at the moment:
[image]

I get the same error message as before:
[image]

baggepinnen, 2023-05-11:
what's the full output of `pkg> st -m`? Preferably as text rather than as screenshots, text is *always* preferable ;)

baggepinnen, 2023-05-11:
The issue is likely related to 
https://github.com/andreasvarga/MatrixPencils.jl/blob/460e14a529b7c300dca4917cefcca94e5d0c0ee5/src/lapackutil2.jl#L3

I'll open an issue in MatrixPencils.jl and close this one here

closed 2023-05-11 by baggepinnen

B-LIE, 2023-05-11:
Here it is:
```julia
(@v1.9) pkg> st -m
Status `C:\Users\Bernt\.julia\environments\v1.9\Manifest.toml`
⌅ [c3fe647b] AbstractAlgebra v0.29.4
  [621f4979] AbstractFFTs v1.3.1
  [1520ce14] AbstractTrees v0.4.4
  [79e6a3ab] Adapt v3.6.1
  [ec485272] ArnoldiMethod v0.2.0
  [4fba245c] ArrayInterface v7.4.3
  [30b0a656] ArrayInterfaceCore v0.1.29
⌃ [4c555306] ArrayLayouts v0.8.18
⌃ [aae01518] BandedMatrices v0.17.18
  [e2ed5e7c] Bijections v0.1.4
  [d1d4a3ce] BitFlags v0.1.7
  [62783981] BitTwiddlingConvenienceFunctions v0.1.5
  [764a87c0] BoundaryValueDiffEq v2.11.0
  [fa961155] CEnum v0.4.2
  [2a0fbf3d] CPUSummary v0.2.2
  [00ebfdb7] CSTParser v3.3.6
  [49dc2e85] Calculus v0.5.1
  [d360d2e6] ChainRulesCore v1.16.0
  [fb6a15b2] CloseOpenIntervals v0.1.12
  [944b1d66] CodecZlib v0.7.1
  [35d6a980] ColorSchemes v3.21.0
  [3da002f7] ColorTypes v0.11.4
  [c3611d14] ColorVectorSpace v0.9.10
  [5ae59095] Colors v0.12.10
  [861a8166] Combinatorics v1.0.2
  [a80b9123] CommonMark v0.8.12
  [38540f10] CommonSolve v0.2.3
  [bbf7d656] CommonSubexpressions v0.3.0
  [34da2185] Compat v4.6.1
  [b0b7db55] ComponentArrays v0.13.12
  [b152e2b5] CompositeTypes v0.1.3
  [f0e56b4a] ConcurrentUtilities v2.2.0
  [187b0558] ConstructionBase v1.5.2
  [d38c429a] Contour v0.6.2
  [a6e380b2] ControlSystems v1.7.2
  [aaaaaaaa] ControlSystemsBase v1.4.2
  [687d7614] ControlSystemsMTK v0.1.12
  [adafc99b] CpuId v0.3.1
  [a8cc5b0e] Crayons v4.1.1
  [717857b8] DSP v0.7.8
  [9a962f9c] DataAPI v1.14.0
  [82cc6244] DataInterpolations v4.0.0
  [864edb3b] DataStructures v0.18.13
  [e2d170a0] DataValueInterfaces v1.0.0
  [bcd4f6db] DelayDiffEq v5.42.0
  [8bb1440f] DelimitedFiles v1.9.1
  [a81e2ce2] DescriptorSystems v1.3.7
  [2b5f629d] DiffEqBase v6.124.0
  [459566f4] DiffEqCallbacks v2.26.1
  [77a26b50] DiffEqNoiseProcess v5.16.0
  [163ba53b] DiffResults v1.1.0
  [b552c78f] DiffRules v1.13.0
  [0c46a032] DifferentialEquations v7.7.0
  [b4f34e82] Distances v0.10.8
  [31c24e10] Distributions v0.25.90
  [ffbed154] DocStringExtensions v0.9.3
  [5b8099bc] DomainSets v0.6.6
  [fa6b7ba4] DualNumbers v0.6.8
  [7c1d4256] DynamicPolynomials v0.4.6
  [4e289a0a] EnumX v1.0.4
  [d4d017d3] ExponentialUtilities v1.24.0
  [e2ba6199] ExprTools v0.1.9
  [c87230d0] FFMPEG v0.4.1
  [7a1cc6ca] FFTW v1.6.0
  [7034ab61] FastBroadcast v0.2.5
  [9aa1b823] FastClosures v0.3.2
[... 249 lines cut ...]
  [1a1011a3] SharedArrays
  [6462fe0b] Sockets
  [2f01184e] SparseArrays
  [10745b16] Statistics v1.9.0
  [4607b0f0] SuiteSparse
  [fa267f1f] TOML v1.0.3
  [a4e569a6] Tar v1.10.0
  [8dfed614] Test
  [cf7118a7] UUIDs
  [4ec0a83e] Unicode
  [e66e0078] CompilerSupportLibraries_jll v1.0.2+0
  [deac9b47] LibCURL_jll v7.84.0+0
  [29816b5a] LibSSH2_jll v1.10.2+0
  [c8ffd9c3] MbedTLS_jll v2.28.2+0
  [14a3606d] MozillaCACerts_jll v2022.10.11
  [4536629a] OpenBLAS_jll v0.3.21+4
  [05823500] OpenLibm_jll v0.8.1+0
  [efcefdf7] PCRE2_jll v10.42.0+0
  [bea87d4a] SuiteSparse_jll v5.10.1+6
  [83775a58] Zlib_jll v1.2.13+0
  [8e850b90] libblastrampoline_jll v5.7.0+0
  [8e850ede] nghttp2_jll v1.48.0+0
  [3f19e933] p7zip_jll v17.4.0+0
Info Packages marked with ⌃ and ⌅ have new versions available, but those with ⌅ are restricted by compatibility constraints from upgrading. To see why use `status --outdated -m`
```

baggepinnen, 2023-05-11:
The issue has been fixed in MatrixPencils and a new release of that package is pending, once merged, you should be able to update to the newly made release and hopefully enjoy a working `minreal` :)