JuliaControl/ControlSystems.jl #857: `ssdata` & storage format of systems?

B-LIE, 2023-06-08:
I'm puzzled by the following:

```julia
> using ControlSystems

> sys = tf([2,1],[5,6,1])

> ssdata(sys)
(nothing, nothing, nothing, nothing)

> ssdata(ss(sys))
([0.0 0.5; -0.4 -1.2], [0.0; 1.0;;], [0.4 0.4], [0.0;;])
```
* This indicates that the system is stored in the format it was created in, and not in state space form?

* Another observation... `ssdata` creates two extra, blank lines for the outputs, `;;` for the C matrix and the D matrix. Why? (I've seen similar things in other Julia code, too, and it puzzles me :-o )

baggepinnen, 2023-06-08:
Why was this puzzling? Transfer functions and statespace models have quite different properties when it comes to computation and it makes sense to be able to represent both.  For one, transfer functions can represent non-proper systems, whereas statespace models (without mass matrix) can not. The computations involved in computing, e.g., frequency responses are also vastly different for the two different representations.

closed 2023-06-08 by baggepinnen

B-LIE, 2023-06-08:
> Why was this puzzling?

"Puzzling" for one with background in MATLAB's ControlToolbox - where `ssdata`  responds with A,B,C,D matrices irrespective of which constructor was used. Also, my impression is that numerically speaking, it is better with storing models in State Space form, so I thought that was done. The documentation of `ssdata` uses generic `sys` as argument. 

But you are right - I shouldn't assume that ControlSystems.jl has exactly the same use as MATLAB's toolbox. 

As you say, storing the system as a transfer function has some advantages in that transfer functions can handle infinite dimensional systems (such as with the `delay`) function. The delay is, of course, a representation of the simple advection equation $\frac{\partial x}{\partial t} = -v\frac{\partial x}{\partial z}$ with $z \in [0,L]$ leading to Laplace transform $\exp{(-\tau s)}$ with $\tau = L/v$. Many other fancy transfer functions can be found which are not fractions of polynomials, such as in various heat exchangers. It would be cool if ControlSystems could allow for such types by multiplying a transfer function with other functions (such as those given by linear PDEs), but there is perhaps not a big "market" for this :-o. 

Anyways, my confusion was due to MATLAB experience.

B-LIE, 2023-06-08:
> I also encourage you to actually look at the source code

I tried to open the ControlSystems.jl source code in GitHub, but didn't really find `ssdata`. 

However, utility function `Base.functionloc` found it. [I've had experience in the past where `Base.functionloc` gives an error message, though.]

baggepinnen, 2023-06-08:
I can recommend the function `dump` to inspect the structure of any composed object
```julia
julia> H = tf(1, [1,1]);

julia> dump(H)
TransferFunction{Continuous, ControlSystemsBase.SisoRational{Int64}}
  matrix: Array{ControlSystemsBase.SisoRational{Int64}}((1, 1))
    1: ControlSystemsBase.SisoRational{Int64}
      num: Polynomials.Polynomial{Int64, :x}
        coeffs: Array{Int64}((1,)) [1]
      den: Polynomials.Polynomial{Int64, :x}
        coeffs: Array{Int64}((2,)) [1, 1]
  timeevol: Continuous Continuous()
```

The macro `@edit` is also useful in locating the source of the code that is being called
```julia
julia> @edit ssdata(H)
```

As a side, we represent time-delay systems as an LFT of a statespace system with a time-delay block:
```
julia> tf(1, [1,1]) * delay(2.0)
DelayLtiSystem{Float64, Float64}

P: StateSpace{Continuous, Float64}
A = 
 -1.0
B = 
 0.0  1.0
C = 
 1.0
 0.0
D = 
 0.0  0.0
 1.0  0.0

Continuous-time state-space model

Delays: [2.0]
```

you can read more about this in the paper https://github.com/JuliaControl/CDC2021/blob/master/CDC2021.pdf

baggepinnen, 2023-06-08:
> Also, my impression is that numerically speaking, it is better with storing models in State Space form, so I thought that was done.

For many things yes, but not for everything, we thus allow the user to choose the representation. We also store `zpk` models as separate arrays of zeros poles and gain, which is preferable in some situations.