JuliaControl/ControlSystems.jl #867: Rendering of state space equations in documentation

albheim, 2023-09-07:
See e.g. [here](https://juliacontrol.github.io/ControlSystems.jl/stable/lib/synthesis/#ControlSystemsBase.add_input), for me the equations for x' and y render on the same line. 
[image]

There are multiple other places that seem to have the same problem.

baggepinnen, 2023-09-07:
There's unfortunately a bit of a conflict between having nice rendering in the docs (latex) and in the REPL (no latex) :( 
In some places, we use code formatting together with unicode to have a bit of a compromise, but for sufficiently complicated things that's just not really feasible. I don't know if something like https://github.com/SimonDanisch/UnicodeFun.jl could be useful to have equations display nice in both places? I've never seen anyone solving this problem before

albheim, 2023-09-07:
Yeah, I don't have a good solution. Maybe some REPL tool that detects markdown math-blocks and just "cleans" them somehow. But I don't really mind that much, I just noted it while looking around and thought I would report it quickly.

closed 2023-09-08 by albheim