JuliaControl/ControlSystems.jl #869: no method matching zpk for step

mzaffalon, 2023-09-08:
I cannot figure out the reason for `step` not working on
```
A2 = [-2.67738     0.0       0.0   0.0;
      0.0        -4.3545     0.0   0.0;
      0.0         0.0        0.0   1.0;
      0.0319049   0.0103997  0.0  -1.83128]

B2 = [64.8148     0.0;
      0.0      236.979;
      0.0        0.0;
      -1.13946   0.0]

C2 = [ 0.0  0.0  0.0   0.0;
       0.0  0.0  3.36  0.0]

D2 = [ 0.0  0.0;
       0.0  0.0]

julia> step(tf(ss(A2,B2,C2,D2)))
ERROR: MethodError: no method matching zpk(::TransferFunction{Continuous, ControlSystemsBase.SisoZpk{T, ComplexF64} where T}, ::Continuous)
```
whereas everything works fine for
```
A = [-2.67738     0.0        0.0        0.0         0.0   0.0;
     0.0         0.0        1.0        0.0         0.0   0.0;
     -0.0716479  12.8845    -0.185185  -0.00255586  0.0   0.0;
     0.0         0.0        0.0       -4.3545      0.0   0.0;
     0.0         0.0        0.0        0.0         0.0   1.0;
     0.0319049  -0.268196   0.0        0.0103997   0.0  -1.83128]

B = [ 64.8148     0.0;
      0.0        0.0;
      0.0        0.280864;
      0.0      236.979;
      0.0        0.0;
      -1.13946    0.0]

C = [ 0.0  3.29  0.0  0.0  0.0   0.0;
      0.0  0.0   0.0  0.0  3.36  0.0]

step(tf(ss(A,B,C,D2)))
``` 
on `Julia v1.9.3` and `[a6e380b2] ControlSystems v1.8.0`.

baggepinnen, 2023-09-08:
Thanks for the bug report, I'll fix it! What's the reason you did the conversion to transfer function? You should generally try to avoid converting a statespace system into a transfer function before simulating, the conversion will suffer from poor numerics, and before actually simulating the transfer function will be converted back into a statespace system.

mzaffalon, 2023-09-08:
Thank you for the prompt reply.

> What's the reason you did the conversion to transfer function?

Ignorance. It is a lab I took in MATLAB and I could not figure out why the linearized plant with controller is unstable whereas Simulink on the full plant gives a stable response.

baggepinnen, 2023-09-08:
[This section of the manual](https://juliacontrol.github.io/ControlSystems.jl/dev/man/numerical/) contains some background on why statespace representations often are to be preferred over transfer functions for numerical reasons.

> I could not figure out why the linearized plant with controller is unstable whereas Simulink on the full plant and the same controller gives a stable response.

Nonlinearities can be both helpful and detrimental. As a simple example of this, most real systems are never going to diverge to infinity, you'd usually hit the nonlinear input saturation at one point or another. Input saturation is usually a bad thing, but here it prevents your system from exploding :)

In your case, it could be that the gain is increased/decreased due to a nonlinearity and that may change the stability properties.

mzaffalon, 2023-09-08:
Thank you for your helpful comment (as usual) and the link!

The states of the Simulink simulation seem to remain close to the operating point where the linearization (using `Simulink` and Julia's `SymPy`) was done, and the non-linear system appears to have rather smooth functions. I have been banging my head over it for a week now.

baggepinnen, 2023-09-08:
How do you simulate the linearized system? Could it be that the sign of the feedback is different in simulink compared to if you close the feedback loop outside of simulink?

PR #870 (2023-09-08): use correct gain type in zpk with length 0 numerator
close #869
commit ce4f79f9 (2023-09-08): close #869
closed 2023-09-08 by baggepinnen