JuliaControl/ControlSystems.jl #873: c2d not working with method input

fibonatic, 2023-09-18:
When I attempt to specify with discretization method I want to use, for example:
`Gd = c2d(G::TransferFunction{<:Continuous}, Ts, method=:zoh)`
I get the error:
`c2d_x0map(::AbstractStateSpace{<:Continuous}, ::Real) got unsupported keyword argument "method"`

baggepinnen, 2023-09-18:
This appears to work for me
```julia
julia> G = tf(1.0, [1, 1])
TransferFunction{Continuous, ControlSystemsBase.SisoRational{Float64}}
   1.0
----------
1.0s + 1.0

Continuous-time transfer function model

julia> c2d(G, 0.1, :zoh)
TransferFunction{Discrete{Float64}, ControlSystemsBase.SisoRational{Float64}}
   0.09516258196404037
-------------------------
1.0z - 0.9048374180359594

Sample Time: 0.1 (seconds)
Discrete-time transfer function model
```

Are you trying to use `method` as a keyword argument? It's a regular positional argument.

baggepinnen, 2023-09-18:
On a side note, are you the user handle `fibonatic` on the control discord? If so, we spoke about coherence functions this morning ;)

fibonatic, 2023-09-18:
Yes, I am currently busy writing some test code, but it has been a while since I used Julia. And I was indeed using `method=:zoh` as input argument. Without the `method=` it is indeed working OK.
Thanks!

baggepinnen, 2023-09-18:
You might have been confused by the fact that the `method` argument hasa default value indicated by the signature in the docstring
```julia
    sysd = c2d(sys::AbstractStateSpace{<:Continuous}, Ts, method=:zoh; w_prewarp=0)
    Gd = c2d(G::TransferFunction{<:Continuous}, Ts, method=:zoh)

closed 2023-09-18 by baggepinnen