JuliaControl/ControlSystems.jl #875: bodeplot does not work well with plotlyjs backend

fibonatic, 2023-09-19:
I am using Julia 1.9.3, ControlSystems v1.8.0 and Plots v1.39.0. The Bode plots are wrong when using the plotlyjs backend.

I used the following code the obtain the attached image:
```
using ControlSystems, Plots
plotlyjs()
bodeplot(tf(1,[1,0.5,1]))
```
[image]

fibonatic, 2023-09-19:
I am able to recreate the same incorrect plot in Julia 1.8.5, ControlSystems v1.7.2 and Plots v1.38.10.

baggepinnen, 2023-09-19:
strange, the plot appears fine here (except for some ugliness in the tick labels) with the same versions of CS and Plots
[image]

how does it look with other plotting backends?

fibonatic, 2023-09-19:
I have tested a couple of different backends (GR, Plotly, PGFPlotsX, UnicodePlots and InspectDR) and all seem to be working OK. I also tried to remove and reinstall Plots, which did not change the plot result when using PlotslyJS. Then I also removed and reinstall PlotslyJS, which did not change the plot result.

baggepinnen, 2023-09-19:
Hmm, I don't think there's anything we can do here then, it must be a bug in either PlotlyJS or the backend code to glue it together with Plots.jl

closed 2023-10-21 by baggepinnen