JuliaControl/ControlSystems.jl #879: `append` on `DelayLtiSystem` does not works

franckgaga, 2023-10-04:
Hi,

Here is a minor bug report. If I do:

```julia
append(tf(1, [10, 1])*delay(3),tf(2, [12, 1])*delay(5))
```
it results in:
```julia
ERROR: type DelayLtiSystem has no field A
Stacktrace:
 [1] getproperty(sys::DelayLtiSystem{Float64, Float64}, s::Symbol)
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/CUXI9/src/types/Lti.jl:70
 [2] (::ControlSystemsBase.var"#116#126")(s::DelayLtiSystem{Float64, Float64})
   @ ControlSystemsBase ./none:0
 [3] iterate(::Base.Generator{Tuple{DelayLtiSystem{Float64, Float64}, DelayLtiSystem{Float64, Float64}}, ControlSystemsBase.var"#116#126"})
   @ Base ./generator.jl:47
 [4] blockdiag(anything::Base.Generator{Tuple{DelayLtiSystem{Float64, Float64}, DelayLtiSystem{Float64, Float64}}, ControlSystemsBase.var"#116#126"})
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/CUXI9/src/connections.jl:209
 [5] append(::DelayLtiSystem{Float64, Float64}, ::DelayLtiSystem{Float64, Float64})
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/CUXI9/src/types/LFTSystem.jl:117
 [6] top-level scope
   @ REPL[52]:1
```
but this works as intended:
```julia
[tf(0) tf(1, [10, 1])*delay(3); tf(0) tf(2, [12, 1])*delay(5)]
```

Thanks !

Francis

closed 2023-10-21 by baggepinnen

franckgaga, 2023-10-22:
Thanks Fredrik!

baggepinnen, 2023-10-22:
Thanks for the report 😊

PR #892 (2023-10-21): Fix `append` for delay system
closes #879
commit 3d4cc676 (2023-10-21): Fix `append` for delay system

closes #879