JuliaControl/ControlSystems.jl #880: OutOfMemoryError on step

mzaffalon, 2023-10-07:
I am not sure whether I should report it here, so apologies if this is the wrong place.
```
A =
[ -6538.4         -40.38          0.0                  0.0;
   409.6               0.0                 0.0                  0.0;
     0.0               0.9164            -28.28       -20.0;
     0.0               0.0                20.0                  0.0]
B =
[ 3846.15    0.0;
    0.0             -288.0;
    0.0                0.0;
    0.0                0.0]
C =[ 0.0  0.0  0.0  1.0]
D =[ 0.0  0.0]
step(ss(A,B,C,D))
```
throws an `OutOfMemoryError` because the temporal step size is too small. Indeed `size(step(ss(A,B,C,D)).x) = (4, 153847, 2)`, a bit of an overkill for this stable slow system.

baggepinnen, 2023-10-12:
The problem here is that this system has one very fast pole, and some very fast poles. When no time-vector is provided, the default time vector tries to be long enough to capture the slow dynamics, but dense enough to capture the fast dynamics. 

With the change in
- https://github.com/JuliaControl/ControlSystems.jl/pull/887

Your step response looks like this
[image]

baggepinnen, 2023-10-12:
This system can be reduced to third order effectively using `balreal`:
```julia
sys = ss(A,B,C,D)
sysr, G, T = balreal(sys)
plot(step.([sys, sysr], 10))
```
the step response is almost identical:
[image]

mzaffalon, 2023-10-12:
Thank you for the help. I am not familiar with `balreal` but it makes zeros and poles even faster:
```
julia> tzeros(sys[1,1])
ComplexF64[]

julia> tzeros(sys[1,2])
1-element Vector{Float64}:
 -6538.461538461539

julia> poles(sys[1,1])
4-element Vector{ComplexF64}:
  -6535.930676480235 + 0.0im
 -2.5308619813034015 + 0.0im
 -14.140000000000017 + 14.144270925007058im
 -14.140000000000017 - 14.144270925007058im

julia> poles(sys[1,2])
4-element Vector{ComplexF64}:
  -6535.930676480235 + 0.0im
 -2.5308619813034015 + 0.0im
 -14.140000000000017 + 14.144270925007058im
 -14.140000000000017 - 14.144270925007058im

julia> sysr, _, _ = balreal(sys);

julia> tzeros(sysr[1,1])
2-element Vector{ComplexF64}:
 3296.0636172361155 + 5676.591154302274im
 3296.0636172361155 - 5676.591154302274im

julia> tzeros(sysr[1,2])
2-element Vector{Float64}:
 -111357.47159489866
  110695.92888170818

julia> poles(sysr[1,1])
3-element Vector{ComplexF64}:
 -2.5308619557301677 + 0.0im
 -14.139998503418344 + 14.144270060245073im
 -14.139998503418344 - 14.144270060245073im

julia> poles(sysr[1,2])
3-element Vector{ComplexF64}:
 -2.5308619557301677 + 0.0im
 -14.139998503418344 + 14.144270060245073im
 -14.139998503418344 - 14.144270060245073im
```

closed 2023-10-12 by baggepinnen

mzaffalon, 2023-10-12:
Oh, I understand what you mean: the fast pole is removed.
```
julia> poles(sys)
4-element Vector{ComplexF64}:
  -6535.930676480235 + 0.0im
 -2.5308619813034015 + 0.0im
 -14.140000000000017 + 14.144270925007058im
 -14.140000000000017 - 14.144270925007058im

julia> poles(sysr)
3-element Vector{ComplexF64}:
 -2.5308619557301677 + 0.0im
 -14.139998503418344 + 14.144270060245073im
 -14.139998503418344 - 14.144270060245073im
```

baggepinnen, 2023-10-12:
Yeah, the fast pole wasn't observable
```julia
julia> gram(sys, :o)
4×4 Matrix{Float64}:
 1.48543e-6  2.37117e-5   6.53234e-5  4.98637e-5
 2.37117e-5  0.000378507  0.00104483  0.000799158
 6.53234e-5  0.00104483   0.0176803   0.025
 4.98637e-5  0.000799158  0.025       0.0530303

julia> gram(sys, :o) |> svdvals
4-element Vector{Float64}:
 0.06599408956924761
 0.004809087557537785
 0.0002874945673967246
 3.685244848996841e-20
```

PR #887 (2023-10-12): Upper-bound length of default time vector
closes #880
commit ce31629c (2023-10-12): Upper-bound length of default time vector

closes #880