JuliaControl/ControlSystems.jl #885: named transfer functions

mzaffalon, 2023-10-12:
Would you consider having named transfer functions and perhaps even moving named_ss from `RobustAndOptimalControl.jl` to `ControlSystems.jl`?

I am asking because the `bodeplot` gives generic names to the outputs and because for LTISystems, the [default name is empty](https://github.com/JuliaControl/ControlSystems.jl/blob/29abee498750849929831ceb053c4a41f7a98845/lib/ControlSystemsBase/src/types/Lti.jl#L101)

baggepinnen, 2023-10-12:
I have considered implementing a new StateSpace type that includes a lot of the functionality that is currently scattered across multiple different statespace types, such as 
- Named signals
- Delays
- Uncertainties
- Static nonlinearities in feedback path

Unfortunately, moving the implementation of a function from ROC to CS cannot be done in a non-breaking way if we want to export the function from CS. I'm also hesitant bringing all that code into ControlSystems, the NamedStateSpace type is still quite immature with lots of problems in how it interacts with other system types and functions not handling it properly etc. I'm thus not yet ready to bring it here and signal that it's ready and stabe. RobustAndOptimalControl is still at version 0.x, which accurately signals that it's not quite as mature as ContorlSystems is. 

What exactly is the problem with `bodeplot`, you should be able to set the names yourself using the regular Plots keywords?

mzaffalon, 2023-10-12:
I understand.

No particular application in mind except for having the plots automatically add the label.

I also liked the ease with which one can build complicated connections used `named_ss` and wondered why one has to import an extra package for something that would ideally belong to `ControlSystems.jl` or even `ControlSystemsBase.jl`.

baggepinnen, 2023-10-12:
Yeah, the reason there is simply that I don't want to put experimental and immature features in a package that aims at being stable and mature :) 

Control engineers are often risk averse (for good reasons, we don't want planes falling out of the sky), so trying hard to signal stability and maturity is especially important for CS.jl.

baggepinnen, 2023-10-12:
I should mention that there is nothing preventing you from implementing a package NamedTransferFuncitons.jl, as long as you extend the appropriate functions
- `ControlSystemsBase.state_names`
- `ControlSystemsBase.input_names`
- `ControlSystemsBase.output_names`

The bodeplot will work as you want