JuliaControl/ControlSystems.jl #897: `pid` with zero proportional gain

mzaffalon, 2023-10-24:
`pid` fails when the proportional gain is 0. This is because the coefficients are [converted to the standard form](https://github.com/JuliaControl/ControlSystems.jl/blob/master/lib/ControlSystemsBase/src/pid_design.jl#L51) which assumes a non-zero proportional gain. 

```
julia> pid(0, 1) # expected 1/s
TransferFunction{Continuous, ControlSystemsBase.SisoRational{Float64}}
0.0
---
1.0
```

albheim, 2023-10-24:
Hmm, that's not good. 

I think using the standard form as the "default" that was used in conversion was something I just picked, didn't really think it through. Probably would have been better to have the parallel form for this since the parameters are not tied together and thus won't (?) lead to these problems.
On the whole that code is not too pretty, and maybe adding another special case would be the simplest for now.

baggepinnen, 2023-10-24:
Yeah, using parallel form by default is probably better in general

mzaffalon, 2023-10-24:
> Anyone want to have a look? Otherwise I could probably have a go during the week.

Let me have a go.

closed 2023-11-21 by baggepinnen