JuliaControl/ControlSystems.jl #919: Change input/output plot order

scls19fr, 2024-03-05:
Hello,

I think it will be more "logical" to first plot input and after output by default when both plotu and ploty are true

[image]

Code to reproduce:
```julia
using ControlSystems
using Plots
τ = 2.0  # s (time constant)
K = 1
s = tf("s")
T1 = K / (1 + τ * s)
t = 0:0.01:10
t0 = 1.0
res = lsim(T1, u, t)
plot(res, label=["output" "input"], plotu=true, ploty=true)
```

Called code
https://github.com/JuliaControl/ControlSystems.jl/blob/0d825400f1fd0126a4ccaa63408ba2b38351d47a/lib/ControlSystemsBase/src/plotting.jl#L105

Kind regards

scls19fr, 2024-03-05:
I also wonder if `lsim` couldn't accept `u` as `Vector{Float}` for single input system like that

Something like

    u = 2.0 * (t . >= t0)

may be enough

baggepinnen, 2024-03-05:
The default is to not plot the input, so if the user requests this, it makes sense to add this plot after the existing one. Also, this is really a trivial thing and I see no point in changing this behavior and confuse existing users.

If you prefer a different order. you can easily override the default using
```julia
plot(step(P, 10), plotu=true, sp=[2 1])
```

closed 2024-03-05 by baggepinnen