JuliaControl/ControlSystems.jl #920: Step response (at a t0 instant) of a first order system with delay doesn't work correctly with lsim (but works fine with step but with t0=0)

scls19fr, 2024-03-05:
Hello,

```julia
using ControlSystems
using Plots
τ = 2.0  # s (time constant)
td = 0.5  # s (delay)
K = 1
T1 = K / (1 + τ * s) * delay(td)
plot(rep, label=["output" "input"], plotu=true, ploty=true)
ylims!(-0.15, 1.15)
```

plots

[image]

but

```julia
t = 0:0.01:10
t0 = 1.0
u = 2.0 * (t .>= t0)'
res = lsim(T1, u, t)
plot(res, label=["output" "input"], plotu=true, ploty=true)
```

raises

```
No methods were found for the model function passed to the equation solver.

The function `f` needs to have dispatches, for example, for an ODEProblem

`f` must define either `f(u,p,t)` or `f(du,u,p,t)`. For more information

on how the model function `f` should be defined, consult the docstring for

the appropriate `AbstractSciMLFunction`.

...
```

mzaffalon, 2024-03-05:
`u` must a function of the state `x` and time `t`: https://juliacontrol.github.io/ControlSystems.jl/stable/lib/timefreqresponse/#ControlSystemsBase.lsim-Tuple{AbstractStateSpace,%20AbstractVecOrMat,%20AbstractVector}

scls19fr, 2024-03-05:
u = (x, t) -> (t < t0 ? 0 : 1)

this is function of the state `x` and time `t` but it raises now

    u must be a vector of size (1,)

Sorry for my questions!

scls19fr, 2024-03-05:
u = (x, t) -> (t < t0 ? [0] : [1])

works fine but is quite tricky (especially when dealing with SISO systems

closed 2024-03-05 by scls19fr

baggepinnen, 2024-03-05:
Try
```
u = (x, t) -> [t < t0]
```

scls19fr, 2024-03-05:
Thanks

    u = (x, t) -> 2 * [t > t0]

works fine