JuliaControl/ControlSystems.jl #925: B matrix cannot contain too many zero elements for baltrunc

Jonacool, 2024-05-10:
During balanced truncation of a decay chain model I found certain errors depending on my choice of B. If it contains to many zeros, then I get an bounds error: 

> ERROR: LoadError: BoundsError: attempt to access 1-element Vector{BigFloat} at index [1:9]
> Stacktrace:
>  [1] throw_boundserror(A::Vector{BigFloat}, I::Tuple{UnitRange{Int64}})
>    @ Base ./abstractarray.jl:737
>  [2] checkbounds
>    @ ./abstractarray.jl:702 [inlined]
>  [3] getindex
>    @ ./array.jl:973 [inlined]
>  [4] baltrunc(sys::StateSpace{Continuous, BigFloat}; atol::Float64, rtol::Float64, n::Int64, residual::Bool)
>    @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/98Jcs/src/matrix_comps.jl:638
>  [5] top-level scope
>    @ ~/Documents/Master-AppliedPhysics/MEP/programming/BT.jl:19
> in expression starting at /home/jonathan/Documents/Master-AppliedPhysics/MEP/programming/BT.jl:19

The full code is:
```
using ControlSystems
using HDF5
using GenericLinearAlgebra
setprecision(BigFloat, 256)

data = h5open("test_chain.h5", "r") do file
    A = read(file["A"])
    B = read(file["B"])
    C = read(file["C"])
    D = read(file["D"])
    (A, B, C, D)
end
A = data[1]
A = BigFloat.(A)
B = data[2][1, :]
#B += 0.0000000001 * rand(50, 1) #workaround
C = data[3]
sys = ss(A, B, C, 0)
sys_red, _, _ = baltrunc(sys, n=9) #Line 19
println(sys_red.A)
```
As I generate A,B,C using python I will describe them. A is a 50x50 sparse matrix with negatives on the diagonal and same values, but positives just right next to it. B should be the initial condition, which is only element 1 being 1 and the other 49 zero. C is a 50x50 identity matrix. 

I found a workaround, by adding a tiny value to every element of B, but I do not feel comfortable doing that. Why does the code 'need' the nonzero elements?

baggepinnen, 2024-05-10:
Could you provide the matrix data ro reproduce the error? It will be hard for me to solve it without being able to reproduce it

baggepinnen, 2024-05-10:
I think the problem is that you request `n=9` modes, but the internal call to `balreal` returns only a single mode, i.e., your system may already be non-minimal. `minreal` should produce a minimal realization before model-order reduction.

You can also try calling `balance_statespace` before calling `baltrunc`, which might help with a poorly balanced `A` matrix.

Jonacool, 2024-05-10:
After calling the `sys = sminreal(sys)` command before `baltrunc` I still get the same error...

baggepinnen, 2024-05-10:
Your system is structurally non-minimal
```
julia> sminreal(sys).nx
1
```
only the first output/state variable is ever affected by the input. Why do you set `B` to the initial condition of a simulation? Are you trying to simulate an impulse response? If so, set `B = I`, perform the balanced truncation and then call `impulse`.

baggepinnen, 2024-05-10:
I'll close this issue since I don't believe there's anything wrong in the implementation. #926 adds a more friendly error message. 

Please feel free to continue the discussion here until you have found a solution to your original problem. To help you, a bit more context would be useful, what is your final goal, what does the model represent and where is it coming from?

closed 2024-05-10 by baggepinnen

Jonacool, 2024-05-13:
Thanks for you help!

I'm trying to make a reduced order model of a fuel burnup of a nuclear reactor. Before I tried the proper orthogonal decomposition, however the timescales in a nuclear reactor vary wildly (10^-20 - 10^40), so it regularly creates unstable models. That is why I'm trying balanced truncation in Julia, with BigFloats. My goal with the decay chain is just to test the methodology, so I'm mainly interested in the impulse response, however in general I would need to make a model that can handle all kinds of initial conditions. 

If I set B to I, the accuracy of the model seems to suffer and it still gives a similar error to the original error if I set the difference between the decay constants large enough (40 orders of magnitude), only it says that it is trying to access an 8 element vector.

baggepinnen, 2024-05-13:
Do you need to reduce the complete model all at once, or could you keep your time scales separate and reduce the fast and slow parts of the model separately before combining?

You can try this PR branch and see if you can retain more modes than 8
https://github.com/JuliaControl/ControlSystems.jl/pull/927

baggepinnen, 2024-05-13:
I think you can add the branch using
```julia
pkg> add ControlSystemsBase#balrealeps
```
and then restart julia before trying

baggepinnen, 2024-05-16:
Did you perform the computations above using BigFloats?

The model-reduction methods in this package all assume an input-output model, but this does not appear to fit your use case, you don't have any inputs?

Jonacool, 2024-05-17:
The simulation is done in Python float64 using the scipy ivp solver with BDF method. All the calculations in Julia are done using BigFloats.

I'm not entirely sure what my inputs must be, as I'm quite new to control theory. I have two models I'm trying to simulate:
1. A decay chain model, whereby the initial condition is a certain amount of the first nuclide which then decays along a chain of nuclides. I guess the only real input would be the amount of the first nuclide you choose. However, then I get the errors with which I started, so I chose the input to be an identity matrix.
2. The burnup of a nuclear molten salt reactor. Initially there is a fuel mixture of <10 nuclides. Afterwards the state evolution is made up by decay and by nuclear interactions dependent on neutron flux. For this model I'm not sure how to exactly fit it to control theory. I'm not sure whether to use the initial fuel mixture as input or the neutron flux.

Thanks again for all your help up until now. It means a lot to me :)

baggepinnen, 2024-05-29:
> The simulation is done in Python float64

You could perform the simulation in BigFloat in julia to see if that resolves some of these issues.

PR #926 (2024-05-10): more helpful error message in baltrunc
closes #925