JuliaControl/ControlSystems.jl #931: Optimization-based tuning–PID controller with Delay

caxelrud, 2024-08-21:
I tried the example of **optimized-based tuning PID** from the documentation at:
https://docs.sciml.ai/ControlSystems/stable/examples/automatic_differentiation/
I modified the process transfer function to one that includes dead-time as:` P = tf(1, [4, 1.0])*delay(1)` .
I got the following error when try to execute:  `res = solve(prob, solver)`.
`ERROR: The advanced interface to the function feedback (with connection keyword arguments) is currently not supported for LFT systems (such as nonlinear and time-delay systems)`

From the documentation, at **Properties of delay systems**, I see examples of the feedback function with process with time-delay.
Let me know if it is a bug or a current limitation.
Thanks!

baggepinnen, 2024-08-22:
You can create feedback connections with delay systems, but not using the advanced version of the `feedack` functions that is used inside the `extended_gangoffour` function, this is a current limitation. You could reformulate the problem to work on discrete-time systems instead, and discretize the delay exactly. To do this, call `c2d` on your system model before the optimization, pass the same sample time `Ts` to `pid` in the `systemspid` function. After that, all computations should be performed on a non-delayed discrete-time model with internal state representing the delay instead

caxelrud, 2024-08-22:
Thanks baggepinnen!
It worked with the instructions you recommended.

baggepinnen, 2024-08-31:
Great!

closed 2024-08-31 by baggepinnen