JuliaControl/ControlSystems.jl #938: Revisiting old code: was there ever an `LTISystem` composed type in `ControlSystems.jl`?

B-LIE, 2024-10-02:
I'm revisiting some Julia code I worked with during early Covid (first half of 2020). In my code, I import `ControlSystems.jl`, and create a struct as:
```julia
struct MFD{T} <: LTISystem
    N::Matrix{Poly{T}}
    D::Matrix{Poly{T}}
...
end
```
My code doesn't work anymore. That is ok, but I'm just wondering whether `LTISystem` was part of `ControlSystems.jl` back then, and whether that type has been moved to some other package (or replaced with something else).

baggepinnen, 2024-10-02:
I believe it's in ControlSystemsBase

B-LIE, 2024-10-03:
Thanks -- I found it. It is exported from ControlSystemsBase, and is defined as:
```julia
abstract type LTISystem{TE<:TimeEvolution} <: AbstractSystem end
```

I suspect it has changed since I used it 4 1/2 years ago, in that it seems to require that parameters are of type TimeEvolution. I used it for transfer matrices/matrix fraction descripton.

closed 2024-10-03 by baggepinnen