JuliaControl/ControlSystems.jl #941: `margin` throws an error when the system has delay

ronisbr, 2024-10-24:
Consider the following system:

```julia
julia> C = 0.6 + 30 * tf([1, 0], [1])

julia> G = tf([0.04, 0.0001, 1.1], [1, 0.03, 254.9])

julia> H = tf([0.25], [1, 1, 0.25])

julia> L = C * G * H * delay(1)
```

If I try to obtain the margins using `allMargins = false`, I get the following error:

```julia
julia> margin(L)
ERROR: BoundsError: attempt to access 1-element Vector{Float64} at index [3]
Stacktrace:
 [1] throw_boundserror(A::Vector{Float64}, I::Tuple{Int64})
   @ Base ./essentials.jl:14
 [2] getindex
   @ ./essentials.jl:916 [inlined]
 [3] sisomargin(sys::DelayLtiSystem{Float64, Float64}, w::Vector{Float64}; full::Bool, allMargins::Bool)
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/kvrre/src/analysis.jl:463
 [4] sisomargin
   @ ~/.julia/packages/ControlSystemsBase/kvrre/src/analysis.jl:444 [inlined]
 [5] margin(sys::DelayLtiSystem{Float64, Float64}, w::Vector{Float64}; full::Bool, allMargins::Bool)
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/kvrre/src/analysis.jl:429
 [6] margin
   @ ~/.julia/packages/ControlSystemsBase/kvrre/src/analysis.jl:410 [inlined]
 [7] margin(system::DelayLtiSystem{Float64, Float64})
   @ ControlSystemsBase ~/.julia/packages/ControlSystemsBase/kvrre/src/analysis.jl:485
 [8] top-level scope
   @ REPL[12]:1
```

If works if I remove the delay or if I set `allMargins = true`:

```julia
```

baggepinnen, 2024-10-24:
Thanks for the issue, it should be fixed by
- #942

closed 2024-10-24 by baggepinnen

ronisbr, 2024-10-24:
Awesome! Thanks for the quick response @baggepinnen !

commit b69a532e (2024-10-24): Merge pull request #942 from JuliaControl/marginfix2

fix #941
commit edd7e299 (2024-10-24): fix #941