JuliaControl/ControlSystems.jl #943: augstate function for full-state feedback functionality.

garazha-ilya, 2024-11-20:
I want to create PR for adding MATLAB-like function [augstate](https://www.mathworks.com/help/control/ref/statespacemodel.augstate.html). It is helpful in scenarios like [this](https://www.mathworks.com/help/control/ug/dc-motor-control.html).

baggepinnen, 2024-11-20:
We already have the function add_output that does this

closed 2024-11-20 by baggepinnen

garazha-ilya, 2024-11-20:
I tried to use non-Latex documentation, but it was weird. I tried using Documenter to add Latex documentation to a webpage, but not to the source code. However, Documenter created a lot of errors.

garazha-ilya, 2024-11-20:
> We already have the function add_output that does this

Could you please add a reference to "augstate" in the documentation so that when you search the Internet for "augsatate julia", the link would lead to this function?

baggepinnen, 2024-11-20:
https://github.com/JuliaControl/ControlSystems.jl/commit/b39d0e8697d8396a2235c77860594d7bff7ee2b1