JuliaControl/ControlSystems.jl #982: Circular reasoning with implicit functions

gdalle, 2025-04-15:
While working on #980 I noticed something strange in your use of ImplicitDifferentiation. You seem to be using a circular reasoning like the following:
```julia
forward(x) = implicit(x)
conditions(x, y, z) = ...
const implicit = ImplicitFunction(forward, conditions)
```
Can you explain why this is necessary? Higher-order AD perhaps?

baggepinnen, 2025-04-22:
Which set of methods in particular are circular? I think they all accept different type signatures or have subtly different names, e.g., `are, arec, ared` etc., otherwise the tests would have failed with stack overflow?

gdalle, 2025-05-03:
For instance this chunk:

https://github.com/JuliaControl/ControlSystems.jl/blob/b7a903b413311b22ef326ae11645445f5cf6f0f4/lib/ControlSystemsBase/ext/ControlSystemsBaseImplicitDifferentiationExt.jl#L11-L39

- `implicit_arec` is an `ImplicitFunction` with `forward_arec` as the solver
- `forward_arec` calls `are`
- `are` calls `implicit_arec`

baggepinnen, 2025-05-03:
If it was circular, the tests would fail with stack overflow, but the tests run all these methods without doing so

gdalle, 2025-05-05:
I guess it must be dispatch then 🤷 feel free to close, it just seemed like a weird design / naming decision

baggepinnen, 2025-05-05:
I can't remember the reason that it looks that way, it might be that some method was required to resolve an ambiguity or add a dispatch that wasn't otherwise hit

baggepinnen, 2026-09-08:
Closing with an explanation for future reference.

The pattern is not circular; it is resolved by dispatch on the element type. The extension defines methods like `are(::ContinuousType, A, B, Q, R::AbstractMatrix{<:Dual})` that intercept calls with dual-number inputs and route them through the `ImplicitFunction`. ImplicitDifferentiation's ForwardDiff rule calls the forward solver with the primal values (`x = value.(x_and_dx)` in `ext/ImplicitDifferentiationForwardDiffExt.jl`), so inside `forward_arec` the call `are(Continuous, A, B, Q, R)` receives `R::Matrix{Float64}` and dispatches to the ordinary Riccati solver, not back to the `Dual` method.

The reason for this design is that user code can call `are`, `lqr`, `kalman`, `lyap`, `plyap`, `hinfnorm` and `schur` unchanged with dual inputs, instead of having to switch to separately named implicit versions when differentiating.

closed 2026-09-08 by baggepinnen