JuliaControl/ControlSystems.jl #987: Different behavior of MIMO system construction on Julia 1.12

franckgaga, 2025-04-26:
Hello!

I don't know if it's intended (probably not) but on Julia v1.12, this code:

```julia
sys = [tf(1) tf(2); tf(3) tf(4)]
```

produces:

```julia
2×2 Matrix{TransferFunction{Continuous, ControlSystemsBase.SisoRational{Int64}}}:

...
```

whereas on Julia v1.11 and below the result is:

```julia
TransferFunction{Continuous, ControlSystemsBase.SisoRational{Int64}}

...
```

This is also brings up #975, since a CI test run on the new `pre` channel would have probably catch this.

baggepinnen, 2025-04-27:
Thanks for reporting the issue, I'll try to have a look at it tomorrow. We have historically tested on nightly as well, but nightly was so often broken that the test ended up being useless, so many false positives that we stopped looking at it. When no one looks at the failing tests due to them always failing for the wrong reasons, they're no longer useful.

franckgaga, 2025-04-27:
I had the same experience with nightly, `pre` seems a better compromise with much less false positive.